fairecasoimeme/ZiGate #316: Add build configuration for host-managed PDM to support large network sizes

schrodingersket, 2020-08-10:
With the introduction of support for host-managed PDM, it makes sense to add a build configuration which supports a corresponding high device count. This issue tracks a pull request which adds support for a network of 200 devices (I'm currently trying to push that up to 255, and will update accordingly).

In addition, this pull request removes the E_SL_MSG_SAVE_PDM_RECORD_RESPONSE wait in pdm_host.c. As network size increases, the coordinator will trigger device table saves more and more frequently; for any non-trivial network, this severely degrades performance while the coordinator waits for a confirmation from the host that each PDM record block has been saved before transmitting the next one. This quickly leads to an unresponsive coordinator.

pipiche38, 2021-03-28:
@schrodingersket when you move to 200 devices. Could you kindly tell what Configuration you have been using ? Question  is around the various Tables like Routing Table, Neighbourg table ....

schrodingersket, 2021-06-11:
Closed in favor of implementing in https://github.com/nimbus9inc/ZiGate

closed 2021-06-11 by schrodingersket