fairecasoimeme/ZiGate #319: [SUGGESTION] Assist in enhancing the zigpy-zigate library for feature parity

Hedda, 2020-09-09:
Suggest that you actively assist in coding the downstream zigpy-zigate library project so that it reaches feature parity with ZiGate.

https://github.com/zigpy/zigpy-zigate

Summery; the zigpy-zigate library project needs help from the whole ZiGate community to make it into a more mature library. What is needed is to map all enhanced functions, features and commands that the ZiGate firmware/hardware supports so that the zigpy-zigate library translate them and pass them along back and forth with the common zigpy API used by all radio libraries for zigpy.

zigpy-zigate is a radio library for zigpy which is used as a Zigbee stack abstraction layer in Home Assistant's ZHA integration component implementation that in turn adds native support for multiple Zigbee adapters, including all ZiGate hardware.

https://www.home-assistant.io/integrations/zha/

and:

https://github.com/zigpy/zigpy/

There is also a another proof-of-concept project that might be able to bring direct-attached ZiGate support to Domoticz here:

https://github.com/pipiche38/Domoticz-Zigpy

It should benefit both your ZiGate sales and maybe also your upstream development if there was improved feature parity (complete support for all ZiGate enhanced features) for ZiGate hardware dongles in all applications that will use zigpy-zigate.

Fact is that without enhanced ZiGate support in zigpy-zigate some end-users might prefer buying other hardware than ZiGate.

https://github.com/zigpy/zigpy-zigate/issues

Examples: 

https://github.com/zigpy/zigpy-zigate/issues/42
https://github.com/zigpy/zigpy-zigate/issues/47
https://github.com/zigpy/zigpy-zigate/issues/44
https://github.com/zigpy/zigpy-zigate/issues/43
https://github.com/zigpy/zigpy-zigate/issues/45
https://github.com/zigpy/zigpy-zigate/issues/38

PS: Currently there is only one developer working on zigpy-zigate and have not had too much time to spend on on zigpy.

Hedda, 2020-09-21:
@fairecasoimeme any feedback? Could you maybe try to get the ZiGate community involved in improving zigpy-zigate support?

Hedda, 2020-09-30:
Perhaps @doudz and/or @pipiche38 have input in regards to this as well as better contacts @fairecasoimeme + ZiGate community?

fairecasoimeme, 2020-10-01:
Hi,
I'll try to check that when that will be possible.
There are anothers priorities but I don't forget zigpy which is important for us.
Fred

pipiche38, 2021-10-13:
I think there is not much to make the zigpy-zigate working

pipiche38, 2021-10-13:
@ruimarinho , the zigate in router mode works great. you simply need to use the proper firmware ( https://github.com/badzz/Zigate-Router )

fairecasoimeme, 2021-10-13:
955 / 5000
Résultats de traduction
hello, I think you are confusing a lot of things. I take care of the ZiGate firmware and I have always responded positively to requests for help from @pipiche or @doudz on firmware upgrades.

On ZHA, I have absolutely no knowledge of the zigpy library but I am always ready to help the developers to modify or improve the firmware in the use of Zigpy.

@Doudz and @Pipiche do a remarkable job and use their leisure time to develop these libraries

On ZHA, which I have already used with the ZiGate, I cannot see how a device cannot pair if it is Zigbee 3.0 compatible.
The only problem that was brought up to me is on the ZiGate-Wifi where there are disconnections related to TCP when the devices do not speak enough.

I remind you that Jeedom uses Zigpy-Zigate and that everything is working correctly and that to my knowledge, there are no devices that cannot be paired.