fairecasoimeme/ZiGate #320: Zigate DIN - Perte tu port USB assez régulirement

MiKLM, 2020-09-25:
Bonjour,
Je possède une ZiGate DIN, qui fonctionne sur un raspberry Pi4 avec Jeedom d'installé dessus.
J'utilise pour contrôler la ZiGate le plugin ZiGate.
J'ai réussi à appairer ma ZiGate et a faire quelques associations d'équipements. Cependant assez régulièrement la ZiGate ne communique plus avec Jeedom et j'ai le message d'erreur comme quoi le port n'est pas configuré alors que 5min avant il l'était et tout fonctionnait bien.
Je sollicite donc votre aide, pour m'aider a trouver ce qu'il peut provoquer cela?
Merci

fairecasoimeme, 2020-09-29:
Bonjour,
Pouvez-vous partager les logs de Jeedom svp ?
Fred

MiKLM, 2020-10-01:
Bonjour,

Voici les logs
Mickaël

[2020-10-01 11:40:10][ERROR][zigate](Thread-110) : ZiGate not found
[2020-10-01 11:40:10][ERROR][zigate](Thread-110) : ZiGate has not been found, please check configuration.
[2020-10-01 11:40:10][DEBUG][root](Thread-110) : External command ZIGATE_FAILED_TO_CONNECT : {'message': 'Failed to connect, retry in 38 sec...'}
[2020-10-01 11:40:10][ERROR][zigate](Thread-110) : Failed to connect, retry in 38 sec...
[2020-10-01 11:40:10][INFO][zigate](Thread-101) : Searching ZiGate port
[2020-10-01 11:40:10][ERROR][zigate](Thread-101) : ZiGate not found
[2020-10-01 11:40:10][ERROR][zigate](Thread-101) : ZiGate has not been found, please check configuration.
[2020-10-01 11:40:10][DEBUG][root](Thread-101) : External command ZIGATE_FAILED_TO_CONNECT : {'message': 'Failed to connect, retry in 38 sec...'}
[2020-10-01 11:40:10][ERROR][zigate](Thread-101) : Failed to connect, retry in 38 sec...
[2020-10-01 11:40:11][INFO][zigate](Thread-83) : Searching ZiGate port
[2020-10-01 11:40:11][ERROR][zigate](Thread-83) : ZiGate not found
[2020-10-01 11:40:11][ERROR][zigate](Thread-83) : ZiGate has not been found, please check configuration.
[2020-10-01 11:40:11][DEBUG][root](Thread-83) : External command ZIGATE_FAILED_TO_CONNECT : {'message': 'Failed to connect, retry in 38 sec...'}
[2020-10-01 11:40:11][ERROR][zigate](Thread-83) : Failed to connect, retry in 38 sec...
[2020-10-01 11:40:11][INFO][zigate](Thread-92) : Searching ZiGate port
[2020-10-01 11:40:11][ERROR][zigate](Thread-92) : ZiGate not found
[2020-10-01 11:40:11][ERROR][zigate](Thread-92) : ZiGate has not been found, please check configuration.
[2020-10-01 11:40:11][DEBUG][root](Thread-92) : External command ZIGATE_FAILED_TO_CONNECT : {'message': 'Failed to connect, retry in 38 sec...'}
[2020-10-01 11:40:11][ERROR][zigate](Thread-92) : Failed to connect, retry in 38 sec...
[2020-10-01 11:40:11][DEBUG][root](Thread-1) : Send to jeedom :  {'action': 'ZIGATE_FAILED_TO_CONNECT', 'message': 'Failed to connect, retry in 38 sec...'}
[2020-10-01 11:40:11][DEBUG] : Array (     [action] => ZIGATE_FAILED_TO_CONNECT     [message] => Failed to connect, retry in 38 sec... )
[2020-10-01 11:40:11][DEBUG][root](Thread-1) : Jeedom reply :  {'success': True}
[2020-10-01 11:40:11][DEBUG][root](Thread-1) : Send to jeedom :  {'action': 'ZIGATE_FAILED_TO_CONNECT', 'message': 'Failed to connect, retry in 38 sec...'}
[2020-10-01 11:40:11][DEBUG] : Array (     [action] => ZIGATE_FAILED_TO_CONNECT     [message] => Failed to connect, retry in 38 sec... )
[2020-10-01 11:40:11][DEBUG][root](Thread-1) : Jeedom reply :  {'success': True}
[2020-10-01 11:40:11][DEBUG][root](Thread-1) : Send to jeedom :  {'action': 'ZIGATE_FAILED_TO_CONNECT', 'message': 'Failed to connect, retry in 38 sec...'}
[2020-10-01 11:40:11][DEBUG] : Array (     [action] => ZIGATE_FAILED_TO_CONNECT     [message] => Failed to connect, retry in 38 sec... )
[2020-10-01 11:40:11][DEBUG][root](Thread-1) : Jeedom reply :  {'success': True}
[2020-10-01 11:40:11][DEBUG][root](Thread-1) : Send to jeedom :  {'action': 'ZIGATE_FAILED_TO_CONNECT', 'message': 'Failed to connect, retry in 38 sec...'}
[2020-10-01 11:40:11][DEBUG] : Array (     [action] => ZIGATE_FAILED_TO_CONNECT     [message] => Failed to connect, retry in 38 sec... )
[2020-10-01 11:40:11][DEBUG][root](Thread-1) : Jeedom reply :  {'success': True}
[2020-10-01 11:40:12][WARNING][zigate](Thread-6) : No response after command 0x00b7 (971)
[2020-10-01 11:40:12][DEBUG][root](Thread-6) : {'result': None, 'success': True}
[2020-10-01 11:40:12][DEBUG] : result callZiGate Array (     [result] =>      [success] => 1 )
[2020-10-01 11:40:12][DEBUG][root](Thread-6) : Message received in socket
[2020-10-01 11:40:12][DEBUG][root](Thread-6) : {'action': 'action_move_temperature', 'args': ['080b', 11, 220]}
[2020-10-01 11:40:12][DEBUG][zigate](Thread-6) : REQUEST : 0x00c0 b'\x02\x08\x0b\x01\x0b\x00\xdc\x00\x01'
[2020-10-01 11:40:12][DEBUG][zigate](Thread-6) : Msg to send b'00c000091f02080b010b00dc0001'
[2020-10-01 11:40:12][DEBUG][zigate](Thread-6) : Encoded Msg to send b'010210c0021002191f02120218021b0211021b0210dc0210021103'
[2020-10-01 11:40:12][ERROR][zigate](Thread-6) : Not connected to zigate
[2020-10-01 11:40:12][DEBUG][zigate](Thread-6) : Waiting for status message for command 0x00c0
[2020-10-01 11:40:14][INFO][zigate](Thread-109) : Searching ZiGate port
[2020-10-01 11:40:14][ERROR][zigate](Thread-109) : ZiGate not found
[2020-10-01 11:40:14][ERROR][zigate](Thread-109) : ZiGate has not been found, please check configuration.
[2020-10-01 11:40:14][DEBUG][root](Thread-109) : External command ZIGATE_FAILED_TO_CONNECT : {'message': 'Failed to connect, retry in 38 sec...'}
[2020-10-01 11:40:14][ERROR][zigate](Thread-109) : Failed to connect, retry in 38 sec...
[2020-10-01 11:40:14][INFO][zigate](ZiGate-Listen) : Searching ZiGate port
[2020-10-01 11:40:14][ERROR][zigate](ZiGate-Listen) : ZiGate not found
[2020-10-01 11:40:14][ERROR][zigate](ZiGate-Listen) : ZiGate has not been found, please check configuration.
[2020-10-01 11:40:14][DEBUG][root](ZiGate-Listen) : External command ZIGATE_FAILED_TO_CONNECT : {'message': 'Failed to connect, retry in 38 sec...'}
[2020-10-01 11:40:14][ERROR][zigate](ZiGate-Listen) : Failed to connect, retry in 38 sec...
[2020-10-01 11:40:14][INFO][zigate](Thread-118) : Searching ZiGate port
[... cut ...]
[2020-10-01 11:40:49][DEBUG] : Array (     [action] => ZIGATE_FAILED_TO_CONNECT     [message] => Failed to connect, retry in 38 sec... )
[2020-10-01 11:40:49][DEBUG][root](Thread-1) : Jeedom reply :  {'success': True}
[2020-10-01 11:40:49][DEBUG][root](Thread-1) : Send to jeedom :  {'action': 'ZIGATE_FAILED_TO_CONNECT', 'message': 'Failed to connect, retry in 38 sec...'}
[2020-10-01 11:40:49][DEBUG] : Array (     [action] => ZIGATE_FAILED_TO_CONNECT     [message] => Failed to connect, retry in 38 sec... )
[2020-10-01 11:40:49][DEBUG][root](Thread-1) : Jeedom reply :  {'success': True}

fairecasoimeme, 2020-10-08:
En effet, c'est curieux ... mais côté zigate, il n'y a pas de raison que le driver coupe et se remette ...
Pouvez-vous contrôler le câble USB ou le connecteur micro-usb ?
Si vous avez un Raspberry, l'alimentation est OK ?
Fred

closed 2021-03-03 by fairecasoimeme