fairecasoimeme/ZiGate #322: Procedure propre d'exclusion forcée d'un periph

tcharp38, 2020-10-02:
Bonjour

Le but est de pouvoir forcer l'exclusion d'un periph, sans aller déclencher cette exclusion du périph lui même.
J'étais parti sur l'utilisation de la cmde 0026, mais apres plusieurs lectures il semble que la cmde 004C soit preferable.
Vous confirmez ?

Si oui, j'observe avec un firmware 3.1c que la premiere reponse suite à 004C est un 8035 avec l'event status=E_PDM_SYSTEM_EVENT_PDM_NOT_ENOUGH_SPACE
D'apres la doc, on doit avoit un status 8000, puis un 8048.

Est ce que ca vous parle ?
Qqn a t'il une procedure d'exclusion qui fonctionne ?

Merci d'avance

pipiche38, 2020-10-02:
Attention, il faut que le device soit un device qui ecoute en permanence, car seul le Device peut sortir d'un reseau.
Donc typiquement pour les device sur batterie, ce sera aléatoire ou il faudra eventuellement une intervention de l'utilisateur pour mettre le device en mode ecoute .

tcharp38, 2020-10-02:
Ha ok. Donc aucune possibilité d'exclure qqn de force.

Qu'en est il de la procédure ? 
004C est la bonne cmde à utiliser ?
Le 8035 que je vois fait du sens ?

pipiche38, 2020-11-18:
Here are my recommendation as regards to device removal.

https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/RemoteDevice.md

tcharp38, 2020-11-21:
Thanks.

- Should I use 004C to force main powered eq exclusion ?
-  If a battery powered device is dead, let's say no more battery or no more functional.
 Then you will never be able to force him to leave and it is still known to Zigate and uses an equipement "slot".
 So what's the solution to get zigate forget it forever ?

pipiche38, 2020-11-21:
Per the ZIgbee 3.0 as far as I known there is no way to have the controller removing it. The device itself has to sent a Leave, otherwise the device will be part of the Gateway Trust Center for ever.

If you want to get rid of it, then you have to Erase PDM , but you will loose all devices !

Per standard, a device one paired can sleep for ever and then at a point of time wake up, send a Device Annoucement (Rejoin) and get it back . Without having the network open as the device know already the transaction  key

tcharp38, 2020-11-21:
Ok then. This might be an real issue for those who reached zigate limit in term of devices and get some dead ones.

pipiche38, 2020-11-22:
Yes. And it is then important to properly removed

pipiche38, 2021-01-12:
The only think possible from what I have seen is to remove the device from the Trust Center. Don't know exactly what are the implication/consequences. Will at a point time some garbage collector will be put in place we don't know as this much more NXP Zigbee Stack implementation

pipiche38, 2021-04-08:
Possible path:
https://community.nxp.com/t5/Wireless-Connectivity/JN5169-issue-on-gateway-coordinator-to-remove-its-endevice/m-p/652970
ZPS_vNtSetUsedStatus()
ZPS_vNwkSaveNt()

@fairecasoimeme can we eventuall have a specific API to force End Device to be removed ?

pipiche38, 2021-04-08:
@tcharp38 
> Ok then. This might be an real issue for those who reached zigate limit in term of devices and get some dead ones.

yes, but I think we shouldn't mix , a potential PDM corruption which makes new pairing impossible, and a real physical limitation