fairecasoimeme/ZiGate #323: Commands doc: Device announce/004D, missing "rejoin information" detail

tcharp38, 2020-10-17:
Can you please enhance the doc for "Device Announce" / "rejoin information"
- either with details
- or link/ref to proper chapter of Zigbee standard

Thanks

pipiche38, 2020-10-19:
There is 2 Device Announce type:

1. which doesn't have LQI nor Rejoin value
2. which have a LQI and rejoin Flags

So for the same Message Type: 0x004D we have : 

1. if len(MsgData) == 22 ==> No Join Flag
2. if len(MsgData) == 24 ==> Join Flag

When receiving a Device Annoucement the Rejoin Flag can give us some information

- 0x00 if joined through association 
- 0x01 if joined directly or used orphaning
- 0x02 if was network rejoin

My experience is as follow:

- 0x00 The device was not on the network. Most-likely it has been reset, and all Unbind, Bind , Report, must be redone
- 0x01 The device was on the Network, but change its route the device was not reset
- 0x02, 0x03 The device was on the network and coming back.  Here we can assumed the device was not reset.

tcharp38, 2021-04-13:
@pipiche38 How to interpret the "no rejoin info" case and do you know why I often see 2 consecutives device announce (first with rejoin 00, the 2nd without rejoin info) ?
Thanks

closed 2021-04-20 by fairecasoimeme