fairecasoimeme/ZiGate #324: Msg 0x8011 gives wrong sequence id

doudz, 2020-10-18:
The message 0x8011 gives the wrong sequence id, it gives the sequence id of the ack instead of the sequence id of the sent command
the original seq id is given by the message 0x8000 after sending a command
then the message 0x8011 is supposed to provide the status of the sent command but the seq id doesn't match

example : 
Send: 0x530 b'02a6ec0101000001040000050005000700
data received 0x8000 b'00**d9**05300000' LQI:0
data received 0x8011 b'00a6ec010000**da**' LQI:0

other example :
Send: 0x530 b'02a6ec0101000001040000050003000700'
Send: 0x530 b'02a6ec0101000301040000050104400200'
data received 0x8000 b'00**d5**05300000' LQI:0
data received 0x8000 b'00**d6**05300000' LQI:0
data received 0x8011 b'00a6ec010000**d7**' LQI:0
data received 0x8011 b'a7a6ec010000**d8**' LQI:0

pipiche38, 2020-10-18:
If I'm not wrong 0x0530 doesn't use the ACK mode and so the 0x8011 should not be expected.

I leave @fairecasoimeme  or @badzz to comment

doudz, 2020-10-18:
if true, it's a problem because it was the purpose of https://github.com/fairecasoimeme/ZiGate/issues/239
and if ack mode isn't used why am I getting 0x8011 ?

pipiche38, 2020-10-18:
Are you in full mode raw, or the normal one?
That could explain

doudz, 2020-10-18:
I'm using the raw mode yes

pipiche38, 2020-10-19:
So then forget what I said. This is the behaviour in Normal mode.

fairecasoimeme, 2020-10-20:
Hi,
Could you give your environment ? device, command, cluster.
I try to get the same issue but with no same result

Fred

pipiche38, 2020-10-20:
@fairecasoimeme do you mean tha in RAW Mode we get x8011 for x0530 command ?

fairecasoimeme, 2020-10-20:
No, I don't get the 0x8011 with v3.1d. only when I execute ZDP command.
This is why I need to know the doudz context

doudz, 2020-10-20:
I'm trying to improve zigpy support.
I made test with firmware 3.1d with Home Assistant 0.116.4 
I got msg 0x8011 when during device discovery

pipiche38, 2020-10-20:
Is that on purpose that the command 0x0530 is without Security ?
Could it be related to the Zigbee command itself embedded into the RAW command and which trigger an APS ACK

fairecasoimeme, 2020-10-20:
@doudz, the 0x8011 is not a ACK from your command, It's a ACK from a "route record" message. I think your device is a router and every command you execute a route record message is send (not visible with zigate but with a sniffer) and a ACK 0x8011 is received.

But there is no ACK for 0x0530 command and unfortunately it's not wanted from us and I don't know (because the lib) if we could do something.

doudz, 2020-10-20:
hum, the device is a Xiaomi vibration sensor, so not a router.

not a good news by the way, I was expected an ACK for command  0x0530 and I was really supposing it worked because I got a lot of message 0x8011 on another installation (with a router)

pipiche38, 2020-10-20:
My understanding is that 0x0530 is a full raw and in such it makes sense not to have the 0x8011 (ACk). At least this was my understanding on the normal mode.
However what I have suggested is to get an other API with a not-full Raw mode based on the function ZPS_eAplAfUnicastAckDataReq

fairecasoimeme, 2020-10-21:
@doudz, I think there is a router in your network.
I try to test the RAW function with ZPS_eAplAfUnicastAckDataReq but I don't know if this function permit to manage cluster which is not enable in ZPSconfig

closed 2021-03-03 by fairecasoimeme