fairecasoimeme/ZiGate #328: Plus de communication avec la zigate suite a sa mise jours

Ultchad, 2020-10-25:
Bonjour,

Problème: Suite a la mise a jours de ma zigate, celle-ci ne répond plus aux commandes qui lui sont passé
_Zigate USB - Prolific (bleu)_
_Mise a jours vers la 3.1d depuis une ancien version (3.0?)_

Test:
J'ai mis a jours sous Windows 10 la zigate dans un premier temps avec succès, puis testé avec le TestGUI ou je n'ai qu'une seul ligne lors de la commande version, pas de réponse de la zigate

J'ai testé ensuite sous Linux , mise a jours toujours noté avec succès, et j'ai fait le test avec le module python [zigate](https://github.com/doudz/zigate), la aussi aucune réponse de la zigate

Je ne sais plus quoi testé a ce niveau

D'avance merci pour votre aide

tcharp38, 2020-10-25:
Salut.
Rapidement, le FW 3.1d est déconseillé pour l'instant. On ne l'a pas validé et on a vu qq soucis.
Donc je te suggere 3.1c.

Si tu as fait une mise à jour depuis FW >= 3.0f normalement tout doit bien se passer.

Ultchad, 2020-10-25:
J'ai fait la mise a jours depuis 3.0d je pense.
J'ai testé avec les firmware 3.1c et 3.1b même problème (maj ok, test de communication ko)

tcharp38, 2020-10-25:
Ha, depuis la 3.0d.
Alors on a peut etre une lacune avec le programmateur integré à Abeille qui n'efface pas la PDM je pense.
Je dois creuser ça.
En + de programmer le FW, "erase PDM" est necessaire pour tout FW AVANT la 3.0f.

Dans tous les cas tu devra reappairer tes periphs apres mise à jour du FW et effacement PDM.

Je te suggere pour ne pas rester bloqué, d utiliser le flasher documenté par Zigate qui, lui sait faire un "erase PDM".

pipiche38, 2020-10-25:
Pourquoi déconseiller le firmware 31d ? Quel est où sont les problèmes ?

tcharp38, 2020-10-25:
Les resultats sont mitigés et pas eu le temps de creuser.
De mon cote J'ai eu des soucis avec mes volets Profalux et la encore pas creusé. Je suis juste revenu en arriere.
Je ne mets pas en cause le 3.1d. C"esst peut etre Abeille lui meme qui pose des soucis mais en ce qui concerne le couple 3.1d+Abeille je deconseille aujourd hui.

pipiche38, 2020-10-25:
7 Volets Profalux à la maison RAS.
J'ai un autre utilisateurs qui a des Volets profalux et également des BSO aucune annomalies.

Sur le plugin Domoticz nous sommes pas mal à etre passé sur la 31d (sachant) et ras.

pipiche38, 2020-10-25:
Par contre effectivement pour @Ultchad en venant de la 30d un Erase PDM est necessaire car ce n'est que depuis la 30f qu'il y a une sauvegarde des données lors du flash firmware

Ultchad, 2020-10-25:
J'ai tésté avec le programe Windows:
- Erase EEPROM > Complete EEPROM : sa a pas fonctionné (maj OK, test communication KO)
- Erase EEPROM > PDM Only puis Complete EEPROM : Sa a fonctionné pour la maj 3.0f
- J'ai ensuite fait la maj vers la 3.1c, et sa fonctionne
- Test de communication sous linux: OK

Merci pour l'aide, sur le moment j'avais pas réalisé que la 3.0f faisait un reset de l'EEPROM

closed 2020-10-25 by Ultchad

tcharp38, 2020-10-25:
> 7 Volets Profalux à la maison RAS.
> J'ai un autre utilisateurs qui a des Volets profalux et également des BSO aucune annomalies.
> 
> Sur le plugin Domoticz nous sommes pas mal à etre passé sur la 31d (sachant) et ras.

Merci @pipiche38 
Ce sont de bonnes nouvelles.
La balle est chez nous. Je vais retenter mes volets.. seul periph avec lequel j'ai eu des soucis.

pipiche38, 2020-10-26:
@tcharp38  avec les Volets Profalux, il faut faire attention à une chose. Ne pas faire de bind sur la  télécommande lors d e l'appairage, seul le volet doit etre bindé avec la Zigate, sinon c'est perte de l'ensemble, et Hard reset 

Après je ne vois rien sur la 31d qui pourrait impacter les volets

tcharp38, 2020-11-03:
Effectivement le FW 3.1d fonctionne bien apres qq améliorations dans Abeille.