fairecasoimeme/ZiGate #329: Pizigate (sur Rpi4)+Zigate Admin (Home Assistant) -> mise à jour 3.1d impossible

vini59fr, 2020-10-27:
Bonjour

Je tente en vain de faire la mise à jour du firmware en 3.1d, de mon Pizigate

je lance le service zigate.upgrade_firmware, le backup du firmware existant se fait bien. HA m'affiche "deconnexion" et puis plus rien.
Obligé d'eteindre electriquement mon Pi.
Quand ca redemarre, le firmware est toujours en 3.1c

Je n'avais pas eu de probleme pour le flash en 3.1c avec ce systeme.

vini59fr, 2020-10-27:
j'ai ces 2 lignes d'erreur dans home-assistant.log

2020-10-27 12:19:50 ERROR (SyncWorker_31) [ZiGate Flasher] Unpack failed, length: 0, msg 
NoneType: None
2020-10-27 12:19:50 ERROR (SyncWorker_31) [ZiGate Flasher] Change baudrate failed
NoneType: None

pipiche38, 2020-10-27:
Ce ne serait pas plus simple de voir avec le forum en charge de Home Assistant et Zigate ?

Ici vous êtes sur le GitHub on est géré le firmware de la Zigate

vini59fr, 2020-10-27:
ok, c'est posté chez Zigate

closed 2020-11-23 by fairecasoimeme