fairecasoimeme/ZiGate #332: Message 8701: quel paquet zigbee declenche le message ?

tcharp38, 2020-11-05:
Salut
Je cherche à comprendre l'adresse qui remonte avec le message 8701 pendant l'inclusion d'un periph (ici Xiaomi door sensor).

87010005870000**0400**00 => donc adresse 0x0400

Sauf que côté Wireshark je n'arrive pas à faire la correspondance.
D'autre part l'adresse du periph est 0x0A69b, donc je ne sais pas d'ou sort ce 400.

La trace montre un broadcast (la requete) mais je n'arrive pas à trouver le message qui déclenche le 8701, et du coup cette adresse 0400.
20	1970-01-01 01:00:18	0x0000	Broadcast	ZigBee	111	Many-to-One Route Request, Dst: 0xfffc, Src: 0x0000

Trace complete
[wireshark-export.txt](https://github.com/fairecasoimeme/ZiGate/files/5494341/wireshark-export.txt)

Une idée ?

pipiche38, 2020-11-05:
Dans la mesure ou ni le Status ni le NwkStatus sont à 00, je ne suis pas certain que ce qui est après est correct !

tcharp38, 2020-11-05:
Merci Pipiche.
Je viens juste de réaliser qu'il y a un PR pending sur le sujet 8701.

Mais peux tu me dire comment je fais le lien avec la trace Wireshark ? Quel paquet declenche le 8701 ?

pipiche38, 2020-11-22:
C'est la Stack Zigbee3.0 de NXP qui fait ca ... Donc je doute qu'on puisse établir une relation directe

pipiche38, 2021-04-08:
https://www.nxp.com/docs/en/user-guide/JN-UG-3113.pdf

Section 2.5.2
The ZigBee stack network layer supports a ‘route discovery’ facility which finds the best available route to the destination, when sending a message. A message is normally routed along an already discovered mesh route, if one exists, otherwise the routing node (Router or the Co-ordinator) involved in sending the message initiates a route discovery. Once complete, the message will be sent along the calculated route.