fairecasoimeme/ZiGate #337: Zigate quitte l'inclusion suite action exterieure

tcharp38, 2020-11-18:
J'avoue etre de + en + perplexe vis à vis du comportement de la zigate dans certains cas (certains sont loggué ici), surtout en 3.1d.
Il se passe des choses que je ne comprends pas.
J'avais déja vu ça l'autre jour
- je mets la zigate en inclusion
- je tente de réinclure mes volets profalux
- et je constate que la telecommande profalux (l'action sur) a stoppé le mode inclusion. Elle ne clignote plus. Pourtant c'est sur que la demande d'arret n'a pas été envoyé à la zigate.

Ca vous parle ?
Comment est ce possible ?
Merci

pipiche38, 2020-11-18:
Quelle action as-tu fait sur la télécommande du profalux, car effectivement l'appui sur STOP va fermer le réseau

tcharp38, 2020-11-18:
Ha, ca c'est possible car ca fait partie de la recommandation Profalux.
Ok donc on peut fermer le reseau de l'exterieur sans envoyer d'ordre à la zigate. 
Mais cette derniere ne remonte aucun evenement dans ce sens si on ne l'interroge pas.
On est en ligne ?

pipiche38, 2020-11-18:
Je ne pense pas que ce soit la recommandation Profalux. En fait le STOP permet simplement de vérifier que le Volet a bien été appairé car il va faire un va-et-vient lors de la fermeture du reseau.

Pour moi en faisant le STOP, la remote va envoyer un broadcast à tous les routers (dont la Zigate) avec un permit to join à 0, ce qui ferme le réseau.

tcharp38, 2020-11-18:
Le stop est meme recommandé par @fairecasoimeme 
[image]

mais ca n'est pas ecrit qu'on verifie quoi que ce soit alors.
Bref je sais pas si il a un interet ou pas.

pipiche38, 2020-11-22:
je pense que tu confond recommandé et faisant partie de a procédure Profalux. Et dans ce cas, effectivement le STOP consiste à fermer le reseau pour vérfier par le va-et-vient du volet que celui-ci a bien été inclus.

pipiche38, 2021-03-02:
not a problem . should be close

tcharp38, 2021-03-02:
I agree.

closed 2021-03-02 by tcharp38