fairecasoimeme/ZiGate #338: Change channel on PiZiGate

Kalypox, 2020-11-22:
Hello,

I'm trying to add a Livolo switch to my PiZiGate (with ZigPy/HA) but it doesn't work.
After research, i saw i had to change the channel but I don't find the procedure to set the channel 26 on my PiZiGate.
The only information is here :
https://zigate.fr/documentation/changer-le-canal-zigbee-pour-les-appareils-livolo/
And it's for ZiGate on Windows. I have PiZigate on Debian.

Thanks

doudz, 2020-11-22:
You have to check the doc of zha

doudz, 2020-11-22:
https://www.home-assistant.io/integrations/zha/#defining-zigbee-channel-to-use

Kalypox, 2020-11-22:
I tried this:
[image]

But nothing happens

Kalypox, 2020-11-22:
I also tried this:

zha:
  zigpy_config:
    device:
      path: /dev/serial0
      baudrate: 115200
    network:
      channel: 26
      channels: [15, 26]

Kalypox, 2020-11-22:
I tried a lot of possibilities. This is my swich:
https://www.aliexpress.com/item/32919238619.html
Maybe it's not possible to bind it with PiZiGate

pipiche38, 2020-11-22:
Please can you use the proper forum. This GitHub is to manage Firmware issues !

Kalypox, 2020-11-22:
np, where is the proper forum ?

doudz, 2020-11-22:
Open an issue on home assistant core github

Kalypox, 2020-11-22:
ok thanks

closed 2020-11-22 by Kalypox