fairecasoimeme/ZiGate #340: Feature request : Improve 0x804a

doudz, 2020-12-05:
Could you improve response 0x804a by adding pu8EnergyValuesList in message ?

https://github.com/fairecasoimeme/ZiGate/blob/master/Module%20Radio/Firmware/src/sdk/JN-SW-4170/Components/ZPSAPL/Include/zps_apl_zdp.h#L1042
https://github.com/fairecasoimeme/ZiGate/blob/master/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/app_general_events_handler.c#L788

https://github.com/zigpy/zigpy-zigate/issues/54

pipiche38, 2020-12-05:
What will be the difference with what we got already  and whcih provide 
ZPS_tsAplZdpMgmtNwkUpdateNotify where uint8*  u8EnergyValuesList; is inclued

doudz, 2020-12-05:
well, I have not tried but are you sure ?
according to doc, https://zigate.fr/documentation/commandes-zigate/ Energy list is missing
and looking at the source code, I don't see u8EnergyValuesList
u8EnergyValuesList is declare in the header file but not use, isn't it ?

pipiche38, 2020-12-05:
I'm using it and I can tell you that the Energy level is there

doudz, 2020-12-07:
ok..., what parameter are you using ? 
I tried to ask zigate as target but I don't get any response
I'm trying to mimic this energy scanner https://github.com/zigpy/zigpy-znp/blob/dev/zigpy_znp/tools/energy_scan.py

pipiche38, 2020-12-07:
Here is the command I'm sending is the following, which I think is the same as you tried to do, except that I'm not using ZiGate as the target. My interpretation was that ZiGate need to exchange messages between itself and a remote node in order to compute the Interference Level.

At least this is how I interpret the NXP ApI used. I might be wrong.

NwkScan - 004A bd10 07fff800 03 01 00 0000

0xbd10 is the Targeted  router
0x07fff800 is the Channel mask ['11', '12', '13', '14', '15', '16', '17', '18', '19', '20', '21', '22', '23', '24', '25', '26']
0x03 is the ScanDuration
0x01 is the ScanCount
0x00
0x0000 is the requestor Zigate)

PS/ I don't this this API is used to replace what the ZiGate does when creating a Network and scanning for the best possible channel. In such I don't think we have an API for that as from the NXP stack is a pure internal stuff

closed 2021-04-20 by fairecasoimeme