fairecasoimeme/ZiGate #344: How to know zigate is full ?

tcharp38, 2021-01-04:
Hello.
Happy new year to all.

How to know zigate is full, and the number of registered devices ?

Thanks

tcharp38, 2021-01-10:
Hi guys.
I will rephrase my question.
We know Zigate has some limits in term of directly connected eq & total equipements.
How can we know/detect in which situation we are ?

Thanks.

waltervl, 2021-03-14:
I have what seems to be a full Zigate with only 30 devices connected. https://zigate.fr/forum/topic/zigate-not-pairing-with-end-devices/#postid-2157
What to do?

pipiche38, 2021-04-08:
https://community.nxp.com/t5/Wireless-Connectivity/how-can-we-know-if-the-controlleur-is-full/m-p/1258800#M10994

tcharp38, 2021-04-09:
@pipiche38 
Do you extract some useful infos from the link you added ?
Knowning the zigate "number of devices" status is still unclear to me and I'm afraid several users have this issue without being able to point it.

waltervl, 2021-04-19:
Some progress here to mention?