fairecasoimeme/ZiGate #348: APDU instances are not released

G1K, 2021-01-12:
Firmware latest COORDINATOR (APDU Instances 9) along with PR https://github.com/fairecasoimeme/ZiGate/pull/342 , but sometimes APDUs are not released (further constantly u8getapduuse: 7). Seen by messages 0x8000 . 

```
 zigbee-herdsman:adapter:zigate:driver:log Send command >>>> RawAPSDataRequest 0x0530 <<<< 
  zigbee-herdsman:adapter:zigate:driver:log Payload: { addressMode: 2, targetShortAddress: 10283, sourceEndpoint: 1, destinationEndpoint: 1, profileID: 260, clusterID: 0, securityMode: 2, radius: 30, dataLength: 5, data: <Buffer 10 07 00 01 00> } +403ms
  zigbee-herdsman:adapter:zigate:driver:log ZiGateFrame { msgCodeBytes: <Buffer 05 30>, msgLengthBytes: <Buffer 00 11>, checksumBytes: <Buffer 2f>, msgPayloadBytes: <Buffer 02 28 2b 01 01 00 00 01 04 02 1e 05 10 07 00 01 00>, rssiBytes: <Buffer >, msgLengthOffset: 0 } +0ms
  zigbee-herdsman:adapter:zigate:driver:log <-- send command  <Buffer 01 02 15 30 02 10 11 2f 02 12 28 2b 02 11 02 11 02 10 02 10 02 11 02 14 02 12 1e 02 15 02 00 02 17 02 10 02 11 02 10 03> +1ms
  zigbee-herdsman:adapter:zigate:driver:log DisableResponse: false +0ms
  zigbee-herdsman:adapter:zigate:driver:log --- parseNext  <Buffer 01 80 02 10 02 10 02 1b ba 02 10 76 02 15 30 02 11 76 02 10 02 14 02 17 02 16 02 10 03> +17ms
  zigbee-herdsman:adapter:zigate:driver:frame:log decoded frame >>> <Buffer 01 80 00 00 0b ba 00 76 05 30 01 76 00 04 07 06 00 03> +421ms
  zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 80 00>, msgLengthBytes: <Buffer 00 0b>, checksumBytes: <Buffer ba>, msgPayloadBytes: <Buffer 00 76 05 30 01 76 00 04 07 06>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 } +0ms
  zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> Status 0x8000 <<<< +0ms
  zigbee-herdsman:adapter:zigate:driver:ziGateObject:info --> frame to object { status: 0, sequence: 118, packetType: 1328, requestSent: 1, seqApsNum: 118, PDUM_u8GetNpduUse: 0, u8GetApduUse: 4, PDUM_u8GetMaxNpduUse: 7, u8GetMaxApduUse: 6 } +422ms
  zigbee-herdsman:adapter:zigate:driver:log --- parseNext  <Buffer 01 80 12 02 10 02 1a e9 02 10 02 11 02 11 02 12 28 2b 76 02 11 02 17 02 10 03> +5s
  zigbee-herdsman:adapter:zigate:driver:frame:log decoded frame >>> <Buffer 01 80 12 00 0a e9 00 01 01 02 28 2b 76 01 07 00 03> +5s
  zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 80 12>, msgLengthBytes: <Buffer 00 0a>, checksumBytes: <Buffer e9>, msgPayloadBytes: <Buffer 00 01 01 02 28 2b 76 01 07>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 } +1ms
  zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> APSDataConfirm 0x8012 <<<< +1ms
  zigbee-herdsman:adapter:zigate:driver:ziGateObject:info --> frame to object { status: 0, sourceEndpoint: 1, destinationAddressMode: 2, destinationAddress: 10283, seqNumber: 118, PDUM_u8GetNpduUse: 1, u8GetApduUse: 7 } +5s
  zigbee-herdsman:adapter:zigate:driver:log --- parseNext  <Buffer 01 80 11 02 10 02 18 ed 02 10 28 2b 02 11 02 10 02 10 76 02 10 03> +14ms
  zigbee-herdsman:adapter:zigate:driver:frame:log decoded frame >>> <Buffer 01 80 11 00 08 ed 00 28 2b 01 00 00 76 00 03> +14ms
  zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 80 11>, msgLengthBytes: <Buffer 00 08>, checksumBytes: <Buffer ed>, msgPayloadBytes: <Buffer 00 28 2b 01 00 00 76>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 } +0ms
  zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> APSDataACK 0x8011 <<<< +0ms
  zigbee-herdsman:adapter:zigate:driver:ziGateObject:info --> frame to object { status: 0, destinationAddress: 10283, destinationEndpoint: 1, clusterID: 0, seqNumber: 118 } +14ms
  zigbee-herdsman:adapter:zigate:driver:log --- parseNext  <Buffer 01 80 02 12 02 10 16 c9 02 10 02 11 02 14 02 10 02 10 02 11 02 11 02 12 28 2b 02 12 02 10 02 10 18 02 17 02 11 02 11 02 10 02 10 20 02 13 67 03> +21ms
  zigbee-herdsman:adapter:zigate:driver:frame:log decoded frame >>> <Buffer 01 80 02 00 16 c9 00 01 04 00 00 01 01 02 28 2b 02 00 00 18 07 01 01 00 00 20 03 67 03> +21ms
  zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 80 02>, msgLengthBytes: <Buffer 00 16>, checksumBytes: <Buffer c9>, msgPayloadBytes: <Buffer 00 01 04 00 00 01 01 02 28 2b 02 00 00 18 07 01 01 00 00 20 03>, rssiBytes: <Buffer 67>, msgLengthOffset: -1 } +0ms
  zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> DataIndication 0x8002 <<<< +0ms
  zigbee-herdsman:adapter:zigate:driver:ziGateObject:info --> frame to object { status: 0, profileID: 260, clusterID: 0, sourceEndpoint: 1, destinationEndpoint: 1, sourceAddressMode: 2, sourceAddress: 10283, destinationAddressMode: 2, destinationAddress: 0, payload: <Buffer 18 07 01 01 00 00 20 03> } +21ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"address":10283,"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":7,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":1,"status":0,"dataType":32,"attrData":3}]},"endpoint":1,"linkquality":103,"groupID":null}' +5s
Zigbee2MQTT:debug 2021-01-12 13:28:07: Received Zigbee message from '0x00124b001b80b9a7', type 'readResponse', cluster 'genBasic', data '{"appVersion":3}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2021-01-12 13:28:07: No converter available for 'DIYRuZ_FreePad' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":3}'
  zigbee-herdsman:controller:device:log Interview - got 'applicationVersion' for device '0x00124b001b80b9a7' +5s
  zigbee-herdsman:controller:endpoint Read 0x00124b001b80b9a7/1 genBasic(["stackVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +5s
  zigbee-herdsman:adapter:zigate:adapter:log sendZclFrameToEndpoint: 
  zigbee-herdsman:adapter:zigate:adapter:log  {
  zigbee-herdsman:adapter:zigate:adapter:log   addressMode: 2,
  zigbee-herdsman:adapter:zigate:adapter:log   targetShortAddress: 10283,
  zigbee-herdsman:adapter:zigate:adapter:log   sourceEndpoint: 1,
  zigbee-herdsman:adapter:zigate:adapter:log   destinationEndpoint: 1,
  zigbee-herdsman:adapter:zigate:adapter:log   profileID: 260,
  zigbee-herdsman:adapter:zigate:adapter:log   clusterID: 0,
  zigbee-herdsman:adapter:zigate:adapter:log   securityMode: 2,
  zigbee-herdsman:adapter:zigate:adapter:log   radius: 30,
  zigbee-herdsman:adapter:zigate:adapter:log   dataLength: 5,
  zigbee-herdsman:adapter:zigate:adapter:log   data: <Buffer 10 08 00 02 00>
  zigbee-herdsman:adapter:zigate:adapter:log } +5s
  zigbee-herdsman:adapter:zigate:driver:log Send command >>>> RawAPSDataRequest 0x0530 <<<< 
  zigbee-herdsman:adapter:zigate:driver:log Payload: { addressMode: 2, targetShortAddress: 10283, sourceEndpoint: 1, destinationEndpoint: 1, profileID: 260, clusterID: 0, securityMode: 2, radius: 30, dataLength: 5, data: <Buffer 10 08 00 02 00> } +3ms
  zigbee-herdsman:adapter:zigate:driver:log ZiGateFrame { msgCodeBytes: <Buffer 05 30>, msgLengthBytes: <Buffer 00 11>, checksumBytes: <Buffer 23>, msgPayloadBytes: <Buffer 02 28 2b 01 01 00 00 01 04 02 1e 05 10 08 00 02 00>, rssiBytes: <Buffer >, msgLengthOffset: 0 } +0ms
  zigbee-herdsman:adapter:zigate:driver:log <-- send command  <Buffer 01 02 15 30 02 10 11 23 02 12 28 2b 02 11 02 11 02 10 02 10 02 11 02 14 02 12 1e 02 15 02 00 02 18 02 10 02 12 02 10 03> +1ms
  zigbee-herdsman:adapter:zigate:driver:log DisableResponse: false +0ms
  zigbee-herdsman:adapter:zigate:driver:log --- parseNext  <Buffer 01 80 02 10 02 10 02 1b b4 02 10 77 02 15 30 02 11 77 02 12 02 17 02 17 02 19 02 10 03> +16ms
  zigbee-herdsman:adapter:zigate:driver:frame:log decoded frame >>> <Buffer 01 80 00 00 0b b4 00 77 05 30 01 77 02 07 07 09 00 03> +20ms
  zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 80 00>, msgLengthBytes: <Buffer 00 0b>, checksumBytes: <Buffer b4>, msgPayloadBytes: <Buffer 00 77 05 30 01 77 02 07 07 09>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 } +0ms
  zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> Status 0x8000 <<<< +1ms
  zigbee-herdsman:adapter:zigate:driver:ziGateObject:info --> frame to object { status: 0, sequence: 119, packetType: 1328, requestSent: 1, seqApsNum: 119, PDUM_u8GetNpduUse: 2, u8GetApduUse: 7, PDUM_u8GetMaxNpduUse:  [... cut ...]
Below is a part of the log. This occurs after the RawAPSDataRequest 0x0530. It is also noteworthy that in the messages there was nothing about 5,6 immediately after u8getapduuse: 4 turned out to be 7 and hung in this position. In the 0x8000 messages at the end, I also added PDUM_u8GetMaxNpduUse u8GetMaxApduUse for debugging.

After that, messages 0x9999 with a load of 0x81 often start arriving, sometimes 5-10 times in a row.

This behavior was previously on firmware 3.1 c before the fix https://github.com/fairecasoimeme/ZiGate/commit/42bdd0b3cc601162ce3331adcfc477902f289ea9

But even now, when the maximum APDU is reached, messages 0x8000 with the status A6 are received

pipiche38, 2021-01-14:
Are you able to identify which is not re-releasing the PDU ?
As in fact the Zigbee stack itself is also allocating Pdus when the devices are requesting Default Response

G1K, 2021-01-14:
I can throw off the full log of communication with zigate and sniff when pairing

G1K, 2021-01-14:
Even without pairing, after several restarts the device started sending Data request -> 0x0000 and ZCL identify -> Broadcast
There were no other packages on the network

[image]

Example
```
Data request
0000   63 88 cc 6e 1d 00 00 2b 28 04 4a 73

ZCL Identify
0000   41 88 36 6e 1d ff ff 00 00 08 02 ff ff 2b 28 1d
0010   aa 28 27 40 06 00 e6 fe 9f 03 00 8d 15 00 00 7b
0020   47 96 75 38 80 4e 61 35 66 a6 26 5c 5a b3 da 88

```

ZiGate does not send messages, but after the next request, 0x9999 with status 0x81 began to arrive
By sending the command permit join, I saw the status

```
  zigbee-herdsman:adapter:zigate:driver:log --- parseNext  <Buffer 01 80 02 10 02 10 02 1b 97 02 10 02 10 02 10 14 02 10 02 10 02 10 02 17 02 16 02 19 02 10 03> +19ms
  zigbee-herdsman:adapter:zigate:driver:frame:log decoded frame >>> <Buffer 01 80 00 00 0b 97 00 00 00 14 00 00 00 07 06 09 00 03> +20ms
  zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 80 00>, msgLengthBytes: <Buffer 00 0b>, checksumBytes: <Buffer 97>, msgPayloadBytes: <Buffer 00 00 00 14 00 00 00 07 06 09>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 } +1ms
  zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> Status 0x8000 <<<< +1ms
  zigbee-herdsman:adapter:zigate:driver:ziGateObject:info --> frame to object { status: 0, sequence: 0, packetType: 20, requestSent: 0, seqApsNum: 0, PDUM_u8GetNpduUse: 0, u8GetApduUse: 7, PDUM_u8GetMaxNpduUse: 6, u8GetMaxApduUse: 9 } +20ms

```

```
  zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> ExtendedStatusCallBack 0x9999 <<<< +1ms
  zigbee-herdsman:adapter:zigate:driver:ziGateObject:info --> frame to object { status: 129 } +13s
  zigbee-herdsman:adapter:zigate:driver:log --- parseNext  <Buffer 01 99 99 02 10 02 12 83 81 02 10 03> +1ms
  zigbee-herdsman:adapter:zigate:driver:frame:log decoded frame >>> <Buffer 01 99 99 00 02 83 81 00 03> +1ms
  zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 99 99>, msgLengthBytes: <Buffer 00 02>, checksumBytes: <Buffer 83>, msgPayloadBytes: <Buffer 81>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 } +0ms
  zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> ExtendedStatusCallBack 0x9999 <<<< +1ms
  zigbee-herdsman:adapter:zigate:driver:ziGateObject:info --> frame to object { status: 129 } +1ms
  zigbee-herdsman:adapter:zigate:driver:log --- parseNext  <Buffer 01 99 99 02 10 02 12 83 81 02 10 03> +0ms
  zigbee-herdsman:adapter:zigate:driver:frame:log decoded frame >>> <Buffer 01 99 99 00 02 83 81 00 03> +2ms
  zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 99 99>, msgLengthBytes: <Buffer 00 02>, checksumBytes: <Buffer 83>, msgPayloadBytes: <Buffer 81>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 } +0ms
  zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> ExtendedStatusCallBack 0x9999 <<<< +1ms
  zigbee-herdsman:adapter:zigate:driver:ziGateObject:info --> frame to object { status: 129 } +1ms
  zigbee-herdsman:adapter:zigate:driver:log --- parseNext  <Buffer 01 99 99 02 10 02 12 83 81 02 10 03> +15s
  zigbee-herdsman:adapter:zigate:driver:frame:log decoded frame >>> <Buffer 01 99 99 00 02 83 81 00 03> +15s
  zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 99 99>, msgLengthBytes: <Buffer 00 02>, checksumBytes: <Buffer 83>, msgPayloadBytes: <Buffer 81>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 } +1ms
  zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> ExtendedStatusCallBack 0x9999 <<<< +1ms
  zigbee-herdsman:adapter:zigate:driver:ziGateObject:info --> frame to object { status: 129 } +15s
  zigbee-herdsman:adapter:zigate:driver:log --- parseNext  <Buffer 01 99 99 02 10 02 12 83 81 02 10 03> +0ms
  zigbee-herdsman:adapter:zigate:driver:frame:log decoded frame >>> <Buffer 01 99 99 00 02 83 81 00 03> +0ms
  zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 99 99>, msgLengthBytes: <Buffer 00 02>, checksumBytes: <Buffer 83>, msgPayloadBytes: <Buffer 81>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 } +0ms
  zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> ExtendedStatusCallBack 0x9999 <<<< +1ms
  zigbee-herdsman:adapter:zigate:driver:ziGateObject:info --> frame to object { status: 129 } +1ms
  zigbee-herdsman:adapter:zigate:driver:log --- parseNext  <Buffer 01 99 99 02 10 02 12 83 81 02 10 03> +0ms
  zigbee-herdsman:adapter:zigate:driver:frame:log decoded frame >>> <Buffer 01 99 99 00 02 83 81 00 03> +1ms
  zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 99 99>, msgLengthBytes: <Buffer 00 02>, checksumBytes: <Buffer 83>, msgPayloadBytes: <Buffer 81>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 } +0ms
  zigbee-herdsman:adapter:zigate:driver:log --> parsed frame >>>> ExtendedStatusCallBack 0x9999 <<<< +1ms
  zigbee-herdsman:adapter:zigate:driver:ziGateObject:info --> frame to object { status: 129 } +1ms
  zigbee-herdsman:adapter:zigate:driver:log --- parseNext  <Buffer 01 99 99 02 10 02 12 83 81 02 10 03> +15s
  zigbee-herdsman:adapter:zigate:driver:frame:log decoded frame >>> <Buffer 01 99 99 00 02 83 81 00 03> +15s
  zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 99 99>, msgLengthBytes: <Buffer 00 02>, checksumBytes: <Buffer 83>, msgPayloadBytes: <Buffer 81>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 } +0ms
[... cut ...]
  zigbee-herdsman:adapter:zigate:driver:ziGateObject:info --> frame to object { status: 129 } +1ms
  zigbee-herdsman:adapter:zigate:driver:log --- parseNext  <Buffer 01 99 99 02 10 02 12 83 81 02 10 03> +0ms
  zigbee-herdsman:adapter:zigate:driver:frame:log decoded frame >>> <Buffer 01 99 99 00 02 83 81 00 03> +0ms
  zigbee-herdsman:adapter:zigate:driver:frame:log ZiGateFrame { msgCodeBytes: <Buffer 99 99>, msgLengthBytes: <Buffer 00 02>, checksumBytes: <Buffer 83>, msgPayloadBytes: <Buffer 81>, rssiBytes: <Buffer 00>, msgLengthOffset: -1 } +0ms

```

pipiche38, 2021-03-02:
@fairecasoimeme , I think that the recent changes tested last week shows a correct behaviour on teh Schneider Wiser for instance. Not sure if it will address all cases

pipiche38, 2021-04-08:
pretty good improvement made @fairecasoimeme . do not see further leak on PDU with the recent code

pipiche38, 2021-04-29:
Firmware 31e

closed 2021-04-30 by fairecasoimeme