fairecasoimeme/ZiGate #351: Topology - LQI Management

pipiche38, 2021-03-11:
I'm using 0x00Ae to build a network topology. The algo is starting from the controller and then queering each Neighbour.

From the source code, I feel that the firmware is reporting only 2bits out of the relationship bitfield which from the documentation is on a 3bit field. 

Last, I'm getting some strange information time to time where the Zigate 0x0000 is sibling with some routers . 

Here after it shows the Zigate sibling with several routers, and if I pick one of the router, this is also confirmed in its neighbour table !

```
Node   Node  Relation        Type Deepth  LQI Rx-Idle
0000   56fb     Child  End Device      1  207  Rx-Off
0000   f8d7     Child  End Device      1  159  Rx-Off
0000   50e0     Child      Router      1  161   Rx-On
0000   cd45   Sibling      Router      0   96   Rx-On
0000   5cbe   Sibling      Router      0   93   Rx-On
0000   ad66   Sibling      Router      0  125   Rx-On
0000   50ad   Sibling      Router      0   82   Rx-On
0000   129a   Sibling      Router      0   88   Rx-On
0000   cdd5   Sibling      Router      0  103   Rx-On

cdd5   0000   Sibling Coordinator      0  245   Rx-On
cdd5   129a   Sibling      Router      2  255   Rx-On
cdd5   cd45   Sibling      Router      2  249   Rx-On
cdd5   50ad   Sibling      Router      2  252   Rx-On
cdd5   5cbe   Sibling      Router      1  255   Rx-On
cdd5   50e0   Sibling      Router      2  188   Rx-On
cdd5   780e     Child  End Device      3  252  Rx-Off
cdd5   ad66   Sibling      Router      2  188   Rx-On
```

tcharp38, 2021-03-11:
Hi @pipiche38 
What is supposed to do 00AE compared to 004E ? It's undocumented.

pipiche38, 2021-03-11:
You right, I'll edit the post, I'm refering to 0x004e as 0x00ae doesn't exist

tcharp38, 2021-03-12:
Ok then I'm interested too. I've seen similar things I could not understand like "unknown" relationship.

pipiche38, 2021-03-12:
@tcharp38 not sure how you get unknown relationship as it can only be
0x00 : Parent
0x01: Child
0x02: Sibling
0x03: Previous Child

and the main issue is that we have 3 bits information sent by the ZiGate into a 2 bits one !

I did further investigation and I have indeed some unexpected Unknown, for Permit to Join, where I saw also PermitJoin On while the entire network is on OFF !

tcharp38, 2021-03-12:
If 0 to 3, then 2 bits is enough unless MSB has another meaning. No ?

Regarding 0x03/Previous child. What does it mean ? I don't understand this relation.

Right now I could not tell you where I saw these unknown but sounds quite in line with your observations. Something is wrongly reported.

pipiche38, 2021-03-12:
I updated . 0x03 is None of above (which you might consider as unknown) and 0x04 is former child

tcharp38, 2021-03-12:
Ok then 3 bits. The point you mentioned.
Could you explain to me what is a "former child" ? How is it differente from "child" ?

pipiche38, 2021-03-12:
Those are part of the norm. ( https://zigbeealliance.org/wp-content/uploads/2019/12/docs-05-3474-21-0csg-zigbee-specification.pdf )
From my understand a End Device (ZED) moving from one parent to an other one during a rejoins process and
The previous parent of this ZED shall remove the ZED 5649 as  one  of  its  children  by  changing  the  Relationship  field  of  the  nwkNeighborTable  to  0x04,  “previous  5650 child.”

The  extended  address,  device  type,  RxOnWhenIdle,  and  permit  joining  fields  have  “unknown”  values  which shall be returned where the values are not available

pipiche38, 2021-04-08:
﻿﻿﻿https://community.nxp.com/t5/Wireless-Connectivity/Strange-Neighbourg-table/m-p/1246576/highlight/true#M10899

https://community.nxp.com/t5/Wireless-Connectivity/Suspect-code-in-JN-AN-1216-Zigbee-3-0-IoT-ControlBridge/m-p/1245091

closed 2021-04-08 by pipiche38