fairecasoimeme/ZiGate #353: FW3.1d: configure reporting response too short with Hue

tcharp38, 2021-05-05:
Hi guys,
I'm facing an incomplete response trying to configure battery reporting on Hue motion sensor (SML002).

> [2021-05-05 06:55:17]   sendCmdToZigate(Dest=Abeille1, cmd=0120, len=0017, datas=02A73E0102000100000000010020002100000dacABCD00)
> 
> [2021-05-05 06:55:18] Reçu: "80020012270001040001020102A73E020000180A07003C"
> [2021-05-05 06:55:18] Reçu: "81200008050AA73E020001003C"
> 8120 = msg type
> 0008 = len
> 05 = crc
> 0A = SQN
> A73E = addr
> 02 = EP
> 0001 = clustid
> 00 = ?? <<<< MISSING INFOS THERE
> 3C = LQI
> 
> [2021-05-05 06:55:18] Abeille1, Type=8002/Data indication, Status=00, ProfId=0104, ClustId=0001, SrcEP=02, DestEP=01, SrcAddrMode=02, SrcAddr=A73E, DestAddrMode=02, DestAddr=0000
> [2021-05-05 06:55:18]   Duplication of 8102 and 8000 => dropped
> [2021-05-05 06:55:18] Abeille1, Type=8120/Configure reporting response, SQN=0A, Addr=A73E, EP=02, ClustId=0001, Attr=00, Status=
> 

I've found this issue https://github.com/fairecasoimeme/ZiGate/issues/240 but could not understand if same pb or not.
If "not supported" by Hue, I would expect a status telling it but still with a correct structure.

Any clues ?
Thanks

pipiche38, 2021-05-05:
Can you clarify which firmware version you are using ? But here is how I decode your frame  : 

> 8120 0008 05 0AA73E02000100 3C
MsgType: 8120
Len: 0008
CRC: 05
Payload: 0AA73E02000100
LQI 3C

Decoding Payload (I think that is the payload below firmware 31d , as 31d get also information on the attribute)
Sqn: 0A
SrcNwkId: A73E
SrcEp: 02
Cluster: 0001
Status: 00

FYI, here are the Config Reporting from the HUe HUB
`"0001": {"Attributes": { "0021": {"DataType": "20", "MinInterval":"1C20", "MaxInterval":"1C20", "TimeOut":"0FFF","Change":"00"}}},`

pipiche38, 2021-05-05:
Pour ce qui est de #240 c'est en rapport avec les Philips Hue Bulbs qui effectivement ne supportent pas le Cfg Rpt et nécessite donc du polling direct

tcharp38, 2021-05-05:
Hi Pipiche
FW is 3.1d no doubt but this "corrupted" case is not the usual one.

Examples from the same trace from the same EQ:

> [2021-05-05 06:55:18] Reçu: "81200008050AA73E020001003C"
> [2021-05-05 06:55:20] Reçu: "8120000A8A0DA73E02040200008D3C"

So I'm receiving both formats !

But I realize now that I may get this "too short response" with other equipments too, Sonoff SNZB-02 for instance.
So finally not linked to Hue.

> [2021-05-05 11:05:44] Reçu: "81200008B86370930104020096"
> [2021-05-05 11:05:49] Reçu: "81200008BE687093010001009C"
> 8120 = type
> 0008 = len
> BE = crc
> 68 = sqn
> 7093 = addr
> 01 = ep
> 0001 = clust
> 00 <<<< ????
> 9C = lqi
>

pipiche38, 2021-05-05:
From the code, there are 2 possibilities to get teh Configure Reporting Reponse
E_ZCL_CBET_REPORT_ATTRIBUTES_CONFIGURE_RESPONSE and E_ZCL_CBET_REPORT_INDIVIDUAL_ATTRIBUTES_CONFIGURE_RESPONSE

So in the case you are reporting the byte you are questioning is as I mentioned earlier the status
https://github.com/fairecasoimeme/ZiGate/blob/master/Module%20Radio/Firmware/src/ZiGate/Source/ZigbeeNodeControlBridge/app_zcl_event_handler.c#L661

tcharp38, 2021-05-05:
Ok then 2 questions:
- both cases have a status and seem to be larger than the packet I receive then I can't get the status (sometimes only) in FW31d.
- how to distinguish between those two 8120 message type ? Or is E_ZCL_CBET_REPORT_INDIVIDUAL_ATTRIBUTES_CONFIGURE_RESPONSE another message ?

pipiche38, 2021-05-05:
if the lenght of the payload is 7 bytes, this is E_ZCL_CBET_REPORT_ATTRIBUTES_CONFIGURE_RESPONSE, if the lenght is greater than 7 bytes then it is E_ZCL_CBET_REPORT_INDIVIDUAL_ATTRIBUTES_CONFIGURE_RESPONSE

tcharp38, 2021-05-06:
As a summary, 8120 response can have 2 formats
- if the lenght of the payload is 7 bytes, this is E_ZCL_CBET_REPORT_ATTRIBUTES_CONFIGURE_RESPONSE
- if the lenght is greater than 7 bytes then it is E_ZCL_CBET_REPORT_ATTRIBUTES_CONFIGURE_RESPONSE

I see that even if always "single" attribute report config, returned format is the one or the other, from the same equipment.

Closing this issue but doc require update to clarify this.
Thanks pipiche.

closed 2021-05-06 by tcharp38