fairecasoimeme/ZiGate #355: Not an ZiGate Issue, but for info - Profalux do not provide Relation ship in LQI Management Response

pipiche38, 2021-05-06:
When using the Link Quality request to retreive the Neighbor Table , it looks like Profalux doesn't populate the Relationship information !

I have reported to profalux this matter

tcharp38, 2021-05-06:
@pipiche38 Talking about profalux, do you know if there are FW updates ?
Initial identifaction was done with "location" instead "modelIdentifier". I'm wondering if they changed this.

pipiche38, 2021-05-06:
Last time I spoke with them , this was not in their Roadmap

They do consider that the Model in in attribute 0x0010 where there put Volet or BSO

tcharp38, 2021-05-07:
Thanks
Strange they choose to be so "no standard", like the pb you just reported.

closed 2021-06-24 by pipiche38

pipiche38, 2021-09-01:
Discussed with Profalux, and in fact the relationship set as None is correct. It means they are neighbour without a Parent/child or Sibling relationship.
In other terms they are at the same horizontal level but do not share the same parent.