fairecasoimeme/ZiGate #356: Official recommendations for FW migration

tcharp38, 2021-05-07:
Not an issue but a clarification.

What are your recommendations to migrate from an older FW to 31e ?
Is it supposed to be transparent ?

3.1d has just been a nightmare for lot's of people including me. Forced to reset PDM et restart from scratch.

Moreover what about reversing back to previous version if too many troubles with 3.1e ?
I assume that it is NOT recommended to switch back to older version unless resetting PDM (and therefore reincluding all EQ).

Please clarify official recommendations.
Thanks

pipiche38, 2021-05-07:
I'll leave @fairecasoimeme  to comment, but here is my understanding

you should not have any issue to upgrade from 31a to 31e, and of course from 31d to 31e

Going back and forth between 31e <-> 31d should also works as there is no change in the PDM

Only the move to ZiGate v3.1e (Optimised PDM) would required Erase EEPROM and Erase PDM and no fall back to 31e would be possible.

schrodingersket, 2021-05-07:
What exactly does the Optimized PDM do? It looks like both versions of `3.1e` are built from the same commit; is there a different `.zpsconfig` file that the Optimized PDM variation is built from?

schrodingersket, 2021-05-07:
> indeed both are based on the same 31e code and so functionality, but the OptyPDM has a differnt set of table size in order to follow NXP sizing recommendation. We have also review the APS, nPDU and aPDU size

Got it. Will there now be two versions for each release moving forward (one for the unoptimized PDM structure and one for the optimized)?

pipiche38, 2021-08-25:
yes, that is correct, you should expect always a OptiPDM and a Legacy version of the Zigate V1 firmware. As the ZiGate V2 it is anyhow a different firmware and it is based on a different PDM