fairecasoimeme/ZiGate #360: 8014 automatic message when leaving 'inclusion' mode ?

tcharp38, 2021-05-13:
Hi guys

It looks like we have to "poll" zigate with 0014 to know if still in inclusion mode or not.

Is there a way such message be automatically raised by zigate (event) as soon as it leaves "inclusion" ?
Might be worth, specifically knowing that some devices may end this out of zigate control (i've seen this sometimes during Profalux inclusion).

Thanks

tcharp38, 2021-05-15:
I change the subject to "automatic 8014 generation when network opened or closed".

I've just experimented with Profalux remote control part of zigate network.
R + STOP make to zigate to blink... network opened but no information reported.

pipiche38, 2021-05-16:
Nice to have, but at the end you won't know if the Network is open for pairing, you'll know only that the ZiGate is open.