fairecasoimeme/ZiGate #361: Zigate wifi: partage simultanné possible ?

tcharp38, 2021-05-22:
Salut
Pas un pb mais une question generale.

Un des utilisateurs Jeedom a une zigate Wifi partagée simultanément entre 2 Jeedom.

Est-ce un fonctionnement supporté ?
La zigate saura en temps reel renvoyer les infos de retour vers le bon initiateur sans soucis ?
Merci

pipiche38, 2021-05-22:
This doesn't make sense for me . This has to be handle by the Wifi module and for sure not the ZiGate.

Last I don't understand the purpose of that. Now I can understand a kind of redundancy (like two boxes) talking to the same Wifi-ZiGate, but in that case, you must be in a Master/Slave mode, where only one box is active.

tcharp38, 2021-05-22:
Ok then the question is for Wifi FW.
Does it handle such comm ?
If not for sure there should be only 1 initiator talking to zigate at a time.

I was suprised by this "use model" but.. some zigate customer understood they could do it.

So finally, what is answer to "does WIFI zigate supports simultaneous dual comm ?"

pipiche38, 2021-05-22:
Yes, they could because there is no limitation in the niumber of TCP-IP session with the Wifi-Module, so indeed it somehow works, but for sure not supported

fairecasoimeme, 2021-05-25:
Hi,
Indeed, this is not supported because there are several TCP / IP channels but only one serial channel.
Thus, if there are 2 TCP connections for example, the serial response will be redirected to a single TCP communication.
Which ? by chance

pipiche38, 2021-06-05:
can we close this ?

tcharp38, 2021-06-07:
Yes. We have the answer.
Thanks

closed 2021-06-07 by tcharp38