fairecasoimeme/ZiGate #362: Add Command to Query Transport Key

schrodingersket, 2021-05-28:
When working with Zigbee networks both in development and in production, it's extremely useful to be able to drop a sniffer onto the network to read and decrypt Zigbee messages. Since location proximity is required and obtaining the link key only allows for decryption of existing messages (and can already be sniffed if the sniffer is running during device commissioning), I believe that the security risk introduced by adding a command to view the established Network Transport Key over serial comms is minimal.

I suggest adding a new command pair `0x001a/0x801a` to query the transport key. I've opened an associated pull request with the necessary changes as well.

The `0x001a` message is an empty message, and `0x801a` simply contains a list of `uint8` bytes that comprise the transport key as the message payload.

pipiche38, 2021-05-28:
Maybe I miss something, but from my end this is someone not secure, so if this is about giving sniff possibility, the best is to do a dedicated firmware for that. Giving the possibility to an application layer to get the transport Key is not a good idea from that perspective.

schrodingersket, 2021-06-04:
Any thoughts on the above @pipiche38 or @fairecasoimeme?

fairecasoimeme, 2021-06-10:
Hi,
I think that we can't add new feature on ZiGate v1 because the low memory but we can add the feature on https://github.com/fairecasoimeme/ZiGatev2

schrodingersket, 2021-06-10:
Hmm... I understand. Would you accept a pull request to update the `README` to indicate that this project is in maintenance mode only then? Had I realized that sooner, I wouldn't have inundated you guys with pull requests to support new hardware and features. (:

That said, I use this project fairly extensively at my company for our ZigBee networks, so I've simply been back-porting features that we found ourselves in need of when working with customer installation sites; if it's expected that no more feature development will occur on this project, I'd be happy to fork it to support continued development since we've committed support to our products with JN5168 chips for a while yet, but I don't want to do that unless you're fairly certain that you won't be maintaining this version much further.

fairecasoimeme, 2021-06-11:
yep, you can do a PR. This project with JN5168 is in maintenance mode only.
Thanks for all

schrodingersket, 2021-06-11:
Closing in favor of implementing this feature in https://github.com/nimbus9inc/ZiGate.

closed 2021-06-11 by schrodingersket