fairecasoimeme/ZiGate #364: Manufacturer-Specific Attribute Reporting does not work

schrodingersket, 2021-06-05:
Currently, messages for `E_ZCL_REPORT_ATTRIBUTES`, `E_ZCL_CONFIGURE_REPORTING_RESPONSE`, `E_ZCL_READ_ATTRIBUTES_RESPONSE`, and `E_ZCL_WRITE_ATTRIBUTES_RESPONSE` messages are all passed up to the application via `Data Indication` messages for any non-standard cluster. Since the format of these message types is pretty iron-clad (and there are quite a few devices on market that utilize custom clusters that just report via these messages for sensing things like air quality which were not in the initial Zigbee spec), it makes sense to parse and propagate such messages via normal program flow and send these messages up to the application in order to facilitate integration with new sensor types.

The required changes to the stack to do this are quite small, so there's a fairly disproportionate work to utility ratio for this feature. I will have a pull request with this functionality shortly.

pipiche38, 2021-06-05:
for me it was the strategy we have adopted from 31d (due to memory limitation), no more code development if Data Indication is provided.
with 31e we have gain some memory , we have for instance even removed some code , but we are still not comfortable and even some parameters having been reduced to get such memory.
Adding code just to go through the standard workflow will not provide any additional functionality

schrodingersket, 2021-06-11:
Closing in favor of implementing this feature in https://github.com/nimbus9inc/ZiGate.

closed 2021-06-11 by schrodingersket