fairecasoimeme/ZiGate #366: [HARDWARE REQUEST] ZiGate Ethernet Gateway (for wired networking) with or without PoE (Power over Ethernet) support

Hedda, 2021-06-10:
I would like to request ZiGate Ethernet Gateway (hardware for wired networking) with or without PoE (Power over Ethernet) support.

Can you please consider making ZiGate Ethernet Gateway hardware board/product with a standard RJ-45 (wired network) port.

Perhaps take this opportunity to also upgrade gateway hardware from ESP8266/ESP8285 to ESP32 as that does not add much cost.

If time to market is a factor then maybe checkout WT32-ETH01 module board by Wireless-Tag:

http://www.wireless-tag.com/portfolio/wt32-eth01/

If you would like to give added value and extended usage then also add PoE (Power over Ethernet) support to that board/product.

If time to market is a factor then maybe checkout or Olimex ESP32-POE-ISO or the LILYGO TTGO T-Internet-POE modules/boards:

https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware

http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1307

As a reference check out the design of the open-source hardware Zigbee gateways by Tube (also known as "Tube's Gateways"):

https://github.com/tube0013/tube_gateways

https://www.tubeszb.com/shop/coordinators/2

[image]

[image]

These would serve as flexible but probably a little more expensive alternative to the existing ZiGate WiFi Pack Gateway hardware.

https://zigate.fr/produit/zigatev2-pack-wifi/

https://zigate.fr/produit/module-wifi-v1-3-compatible-zigate/

pipiche38, 2021-06-10:
it would be simpler to contact the zigate.fr , the purpose of this github is to focus on the firmware for ZiGate, and not hardware related

fairecasoimeme, 2021-06-10:
ZiGate-ethernet is on dev. you have to follow me on twitter / facebook or insta

closed 2021-06-10 by fairecasoimeme

fairecasoimeme, 2021-08-13:
Hi @Hedda ,
ZiGate-Ethernet is released and you can find informations here : 
[https://zigate.fr/produit/zigate-ethernet/](https://zigate.fr/produit/zigate-ethernet/)
[https://zigate.fr/documentation/descriptif-de-la-zigate-ethernet/](https://zigate.fr/documentation/descriptif-de-la-zigate-ethernet/)
[https://github.com/fairecasoimeme/ZiGate-Ethernet](https://github.com/fairecasoimeme/ZiGate-Ethernet)
[https://zigate.fr/documentation/installation-de-tasmota-pour-la-zigate-ethernet/](https://zigate.fr/documentation/installation-de-tasmota-pour-la-zigate-ethernet/)

Also you can install ESPHome with [ZbBridge ](https://github.com/thegroove/esphome-zbbridge) (I'll do the doc)

ZiGate-Ethernet is compatible with PiZiGate, Raspbee1/2 and other RPI modules with ZHA.  (I'll do the doc)