fairecasoimeme/ZiGate #367: Mark this repository as unmaintained

schrodingersket, 2021-06-11:
As per @fairecasoimeme's comment in #362, this repo is to be marked as maintenance only.

_Originally posted by @fairecasoimeme in https://github.com/fairecasoimeme/ZiGate/issues/362#issuecomment-859343471_

closed 2021-06-22 by fairecasoimeme

pdecat, 2021-06-24:
@schrodingersket great initiative!

Do you intend to maintain compatibility of your fork with zigpy for use in Home Assistant ?

pipiche38, 2021-06-24:
It has been reported by the zigate.fr manufacturer that is still supporting and will be posting fixes.

pdecat, 2021-06-24:
> It has been reported by the zigate.fr manufacturer that is still supporting and will be posting fixes.

@pipiche38 I meant this question for the new features introduced by the fork.

pipiche38, 2021-06-24:
My understanding is that Zigpy works on RAW mode, so doesn't realy rely on what the firmware is doing

schrodingersket, 2021-06-24:
@pdecat - I don't plan to add a huge number of new features; I expect that most of what is added will be improvements to network topology, configuration changes, and the like. That said, if new features do get brought in (either by me or the community), I'll certainly see what I can do about `zigpy` support. I personally haven't used the library, as I use a custom library, but I can't imagine it'd be too difficult to maintain `zigpy` support moving forward.