fairecasoimeme/ZiGate #372: Not an issue: any way to know current device binding ?

tcharp38, 2021-07-11:
Guys,
Is there a way with zigate to interrogate a device to know its binding ?
Thanks

pipiche38, 2021-07-12:
You might look at Mgmt_Bind_req  from the Zigbee  Alliance ( https://zigbeealliance.org/wp-content/uploads/2019/11/docs-05-3474-21-0csg-zigbee-specification.pdf )

> The Mgmt_Bind_req is generated from a Local Device wishing to retrieve the contents of the Binding Table  from  the  Remote  Device.  The  destination  addressing  on  this  command  shall  be  unicast  only  and  the  destination address must be that of a Primary binding table cache or source device holding its own binding 

Of course it will work for remote which are able to Receive on Idle

tcharp38, 2021-07-12:
Thanks @pipiche38 
I assume that this request has to be generated thru 0530 zigate command. Correct ?

pipiche38, 2021-07-12:
Yes, because as of today we don't have any API to do it from the firmware

tcharp38, 2021-07-12:
@pipiche38 
Did you use this to know default binding of a remote for ex ? I'm trying to apply this on Traffri on/off switch but no success so far.
Looks like there is no binding while there is at least 1 to group.

pipiche38, 2021-07-13:
No, I’m not using it.
I just do a bind if it is needed, if there is already one in place this should be overriding

tcharp38, 2021-07-13:
Ok i made progresses

Binding table from a zigbee curtain switch:

> Abeille1, Type=8002/Data indication, Status=00, ProfId=0000, ClustId=8033, SrcEP=00, DestEP=00, SrcAddrMode=02, SrcAddr=7C4F, DestAddrMode=02, DestAddr=0000
>    Binding table response, SQN=12, Status=00, tableSize=1, index=0, tableCount=1
>    EC1BBDFFFEA898EE, 01, 0000 => group 0000
> 

Binding table from Xiaomi lum (in line with expectations):

>  Abeille1, Type=8002/Data indication, Status=00, ProfId=0000, ClustId=8033, SrcEP=00, DestEP=00, SrcAddrMode=02, SrcAddr=9007, DestAddrMode=02, DestAddr=0000
>    Binding table response, SQN=12, Status=00, tableSize=2, index=0, tableCount=2
>    04CF8CDF3C77164B, 01, 0004 => EP01 @00158D0001ED3365
>    04CF8CDF3C77164B, 01, 0100 => EP01 @00158D0001ED3365
> 

but table from Ikea tradfri is empty. Looks like this device does not support spec properly. The remote controls group 0000 for sure:

> Abeille1, Type=8002/Data indication, Status=00, ProfId=0000, ClustId=8033, SrcEP=00, DestEP=00, SrcAddrMode=02, SrcAddr=F026, DestAddrMode=02, DestAddr=0000
>    Binding table response, SQN=12, Status=00, tableSize=0, index=0, tableCount=0
>

tcharp38, 2021-07-13:
Just to complete information, it looks like binding table can only report binding infos from a "server" cluster to something else.
No informations on "client" clusters but might be normal. Which means no way to detect the default binding of a remote but this is my understanding.

pipiche38, 2021-09-01:
suggest to close that issue

pipiche38, 2021-11-04:
should be close