fairecasoimeme/ZiGate #381: Integrate and support new Aqara Door Sensor MCCGQ14LM

pipiche38, 2021-11-27:
It looks like the new Aqara Door Sensor ( MCCGQ14LM ) does a Node Descriptor toward ZiGate and in that case the ZiGate (Controleur) MUST answer with Xiaomi manufacturer code.

Please do implement such feature as it has been done by the other GW ( z2m, deConz ) so Zigate can support that model 

CC: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5403#issuecomment-980436576

tcharp38, 2021-11-27:
We should receive a 8002 message isn't it ?
Then the plugin could answer.
Am I correct ?

pipiche38, 2021-11-27:
There no reason to get a 8002 as the firmware is processing and responding to this message. Which is the issue I reported. Zigate is answering the Node Descriptor request with a node descriptor response ( which details Zigate info like zigate manuf code) and here we want in that case à Xiaomi manuf code

tcharp38, 2021-11-27:
Ok I missed that.
Zigate is handling response but you need to set a custom response.
Thanks

pipiche38, 2021-11-29:
Here is what have been done for z2m and there is a similar one for deConz
https://github.com/Koenkk/Z-Stack-firmware/blob/f29b4525bb7bfdc28c91139d0926469bb5754ab9/coordinator/Z-Stack_3.x.0/firmware.patch#L533

rcrocus, 2021-12-04:
Hello,
Is that possible to make this evolution on the V1 firmware (not only for the V2) ?
Thanks a lot.

rcrocus, 2021-12-10:
Hello,
After the upgrade of my Zigate v1 to v3.21, it works perfect.
Thanks a lot to everyone.

closed 2021-12-16 by fairecasoimeme