fairecasoimeme/ZiGate #386: Doc: Cmd 0041/IEEE addr resquest

tcharp38, 2021-12-01:
There are 2 addresses in 0041/IEEE address request command

> target short address: uint16_t
> short address: uint16_t

Which one is the device address and what is used the other one for ?
Thanks

pipiche38, 2021-12-01:
Are you sure about that ?
For me 0x0041 is the IEEE address request when we are giving a Short ID

[image]

tcharp38, 2021-12-01:
This is what I'm expecting, giving only the short addr to get the extended one.
But the doc talks about 2 short addr

[image]

pipiche38, 2021-12-01:
most likely a documentation issue

tcharp38, 2021-12-01:
Currently Im putting device short addr in both fields. Seems ok.
How do you use it ?

pipiche38, 2021-12-02:
Target short address and then 0x00 and 0x00 for the two other fields

tcharp38, 2021-12-02:
I don't understand how it can work on your side then.
Doing this leads to requesting address from 0000

[image]

Payload sent to Zigate is in line with your comment
cmd=0041, len=0004, datas=A0CF0000
where A0CF is my device addr for which I need the IEEE.

So sounds to me that there are 2 short adresses but I still don't understand why.

pipiche38, 2021-12-02:
Here is what is in the NXP documentation
```
This function requests the 64-bit IEEE (MAC) address of the node with a particular 
16-bit network address. The function sends an IEEE_addr_req request to the 
relevant node, specified through uDstAddr.
The network address of the node of interest must also be specified in the request, 
represented by the structure below (detailed further in Section 8.2.2.2).
   typedef struct {
      uint16 u16NwkAddrOfInterest;

      uint8  u8RequestType;

      uint8  u8StartIndex;

   } ZPS_tsAplZdpIeeeAddrReq;
The required IEEE address will be received in an IEEE_addr_resp response, which 
should be collected using the function ZQ_bZQueueReceive() and stored in a 
structure of type ZPS_tsAplZdpIeeeAddrRsp (detailed in Section 8.2.3.2). Note 
that this response can optionally contain the IEEE addresses of the responding 
node’s neighbours (this option is selected as part of the request through 
u8RequestType). 
```

Here is how it is called in the firmware

```
u8Status        =  APP_eZdpIeeeAddrReq ( u16TargetAddress, 1st short address
                                                         u16LookupAddress,, 2nd short address
                                                         au8LinkRxBuffer[4], u8RequestType
                                                         au8LinkRxBuffer[5], u8StartIndex
                                                         &u8SeqNum );
```

My understanding is 
TargetAddress is the short address of a Router/Controller to which you will request to do the lookup
LookupAddress is the short address for which you are looking for an IEEE

pipiche38, 2021-12-02:
After few tests, it looks like short address doit etre égal à lookupaddress.
So no clue why, this should be asked to NXP

tcharp38, 2021-12-02:
Thanks @pipiche38 
This is in line with my observation.
Anyway doc must be updated to mention this.