fairecasoimeme/ZiGate #387: Mise à jour PIZIGATE impossible

mickadam29, 2021-12-05:
Bonjour,

Je tente de faire la mise à jour d'une pizigate.
J'obtiens cette erreur à chaque tentative

jeedom@jeedom:~/JennicModuleProgrammer/Build $ ./JennicModuleProgrammer -V 6 -e -P 115200 -f /home/jeedom/ZiGate_v3.1e_OPDM.bin -s /dev/ttyAMA0
JennicModuleProgrammer Version: 0.7(r56124)
Setting baudrate: 115200
Detected Chip: Unknown
MAC Address:   00:15:8D:00:03:EF:8F:BA
Opened firmware file size 208404
Module uses Bootloader v2 Header
Erasing:   100%
Writing Program to Flash
Writing:   100%
Verifying Program in Flash
Verifying:   0%

Error reading at address 0x00000180

Pizigate défectueuse ou mauvaise manip de ma part ?
En mode production, aucun souci. Je souhaite juste m'assurer qu'elle soit en mode OPDM. 
Ce qui me surprend c'est qu'il est impossible de détecter le chipset

Merci de votre aide. 
Je viens d'acheter une Zigate-Ethernet pour exploiter cette Pizigate. Ca me semble compromis

tcharp38, 2021-12-05:
@mickadam29 
Salut. As tu essayé d'Abeille ? Ou est ce une v2 ?

mickadam29, 2021-12-05:
Salut. En principe c’est pas une V2. Mais je me demande comment vérifier ce point. Elle est déjà en 3.1E et je voulais la passer en OPDM. Que ce soit à partir de Abeille ou via l’OS la phase de vérification indique une erreur.

mickadam29, 2021-12-05:
La facture de février 2021 indique Pizigate. Ni V2 ni Pizigate+ . Pas certain qu’elle existait en 02/21

tcharp38, 2021-12-05:
Abeille ne supporte pas les v2/+ mais je viens de verifier ca fonctionne toujours avec ma PiZigate

> Lancement de la programmation du firmware
> - Port tty: /dev/ttyS1
> - Firmware: ZiGate_v3.20-OPDM.bin
> JennicModuleProgrammer Version: 0.7(r56124)+
> - Detected Chip: JN5168
> - MAC Address:   00:15:8D:00:01:EE:02:CB
> Setting baudrate: 115200
> Opened firmware file size 208464
> Module uses Bootloader v2 Header
> Erasing:     0%
> [AErasing:   100%
> Writing Program to Flash
> [AWriting:     0%
> ..
> AVerifying:  99%
> [AVerifying:  99%
> [AVerifying: 100%
> = Success
> 

Tu as quoi comme trace avec Abeille ? Le programmateur est un peu modifié et du coup il reconnait le device v1.

mickadam29, 2021-12-05:
Je vérifie ça et te fais un retour. Chez moi le chipset n’est pas reconnu et ça plante à 1 à 5% de la phase verifying 

> Le 5 déc. 2021 à 22:56, Tcharp38 ***@***.***> a écrit :
> 
> Module uses Bootloader v2 Header
> Erasing

mickadam29, 2021-12-05:
Tu dis Abeille supporte pas mais tu flashes en 3.2 ? Je croyais la 3.2 réservée au V2

> Le 5 déc. 2021 à 23:06, Mickael ADAM ***@***.***> a écrit :
> 
> ﻿Je vérifie ça et te fais un retour. Chez moi le chipset n’est pas reconnu et ça plante à 1 à 5% de la phase verifying 
> 
>> Le 5 déc. 2021 à 22:56, Tcharp38 ***@***.***> a écrit :
>> 
>> Module uses Bootloader v2 Header
>> Erasing

tcharp38, 2021-12-06:
3.2 n'est pas suffisant. Il faut voir les autres chiffres aussi
en v1: tu devrais avoir 0003-0320 (legacy) ou 0004-0320 (OPDM)
en v2: je crois que c'est 0005-0320

Abeille affiche cette info

[image]

tcharp38, 2021-12-06:
0003-031E = 3.1e legacy
0004-031E = 3.1e OPDM

et oui on peut mettre une 3.20 OPDM sur une v1. Je n'ai que ca

Du coup le device JN5168 est bien reconnu ?  Si oui t 'es bien sur une v1 et ta PiZigate semble avoir un souci sinon je seche.

mickadam29, 2021-12-06:
Le module est un E752G4M20S et donc doit contenir le JN5168. Je vais tenter une 3.2 OPDM dans ce cas.

mickadam29, 2021-12-06:
La 3.20 OPDM est passée. Merci à toi @tcharp38

tcharp38, 2021-12-06:
Comment tu expliques que ca ne fonctionnait pas ?

mickadam29, 2021-12-06:
Probablement pas à jour niveau OS. Beaucoup de paquets mis à jour apt-get update/upgrade. Maintenant je vois de nouveau la mise à jour OTA. J'ai pas tenté autre chose que 320OPDM et c'est passé. Jennic voit bien un JN5168. Par contre impossible de faire le reset PDM. Tous mes objets sont encore pilotés et donc connus de la Pizigate après l'installation de 320 OPDM. Une idée ?

tcharp38, 2021-12-06:
Ouah, la non pas d'idées. Mais la ca devient un pb Abeille. Donc plus sur ce forum.

mickadam29, 2021-12-06:
ok. Je referme. J'ouvre un pb sur Abeille ??

tcharp38, 2021-12-06:
Oui. On va supposer que c est un souci Abeille. Sinon retour.. ici.

closed 2021-12-06 by mickadam29

tcharp38, 2021-12-07:
Je remets une couche sur le sujet "mise à jour".
En effet en jonglant 3.20 vers 3.1E puis à nouveau 3.20 je suis tombé sur le meme probleme à savoir ca plante pendant la phase de vérif. 
Je m'en suis sorti en insistant 2 fois de +, et la c'est passé. 

Bref, à garder en tête mais le programmateur Jennic semble avoir parfois qq faiblesses.