fairecasoimeme/ZiGate #388: Discover attributes (extended) response with FW 3.20

tcharp38, 2021-12-06:
I was investigating why my discovery assistant was failing.
I realized that now, at least with FW 3.20, either you get a 8002 answer that you can properly decode or a 8140/8140 but not both as before.

Summary

Discover attributes extended response
    0B04 => 8002
    0702 => 8141
    0006 => 8141
    0005 => 8141
    0004 => 8141
    0000 => 8141

Discover attributes response
    0B04 => 8002
    0702 => 8140
    0006 => 8140
    0005 => 8140
    0004 => 8140
    0000 => 8140

Can someone confirm such behavior ?
If yes, what is the long term rule ?
Since 8002 is no longer present for all cases, my code is dead.
Thanks

pipiche38, 2021-12-06:
my understanding is that in 
- standard mode you should get 8140 Which is the Attribute Discovery response.
- in hybrid mode you probably get the 2, but never try
As I understood Abeille is using the hybrid mode, you might want to check that you are in hybrid mode and not the standard one

tcharp38, 2021-12-07:
Thanks @pipiche38 
I close this subject as the pb was on my side (no longer in 'hybrid' mode).

closed 2021-12-07 by tcharp38