fairecasoimeme/ZiGate #390: Zigate v1 upgrave de firmware

Jake309, 2021-12-19:
Salut,
j'ai passé la fin de journée à essayer de mettre à jour ma zigate v1.
v3.21 OPDM, légacy, 3.20 OPDM ou légacy rien ne fait, elle ne veut plus rien faire.

En dernière tentative, je l'ai testé avec TestGUI, tout ce passe bien sauf quand je fais Permit Join, j'ai le message suivant:
Type: 0x8012 (Unrecognized), elle clignote bien, mais aucune réaction quand je mets différents équipement en mode inclusion.

Une idée ?

A savoir que j'avais fait une sauvegarde de la flash et l'eeprom (j'étais en 3.1d), et au 1er upgrave, en 3.21 legacy, j'ai restauré mon backup mais ca n'avait pas marché à voir.

Est ce possible qu'elle soit HS ?!

Merci :)

Jake309, 2021-12-19:
Achete le 28/12/2020, elle aura pas eu 1an, cest dommage.

pipiche38, 2021-12-19:
quand on passe d'un firmware legacy à un firmware optPDM il est necessaire de faire un Erase EEPROM.
L'avez-vous fait.
Après il vous faut initialiser la Zigate ... A voir en fonction de l'application que vous utilisez

Jake309, 2021-12-19:
Oui, bien sur, jai fait lerase. 
Jai suivi la procédure sur le site de zigate
Sur jeedom comme sur testgui,  l'initialisation a été faite. 
Je suis revenu au fondamentale, avec testgui, on s'affranchit des plugins etc. On est direct sur la zigate. Mais ça ne pas été un succès non plus

Jake309, 2021-12-20:
J'ai des erreurs dans mon fichier de log après flash, quelqu'un peut m'aider à les comprendre ?
[FlashGUI.exe.log](https://github.com/fairecasoimeme/ZiGate/files/7747085/FlashGUI.exe.log)

tcharp38, 2022-02-11:
@Jake309 
Comment ca s'est terminée cette histoire ? Je crois qu'un autre utilisateur est dans la meme panade que tu l'étais.
Merci

ruimarinho, 2022-02-11:
My story ended the same with my two Zigate v1's now getting dust due to lack of support so I've decided to switch to a https://www.tubeszb.com/ coordinator and router. Couldn't be happier with my choice.

tcharp38, 2022-02-11:
Dommage :(
Merci pour votre feedback.

Jake309, 2022-02-11:
je vais refaire des test tcharp ;)

Jake309, 2022-02-11:
En la retestant après une nouvelle maj, le logiciel testGUI ne donne rien quand je mets des équipements en inclusion... donc même pb.
Par contre, j'ai voulu la mettre sur abeille, et la, pas mal d'erreur qui apparaisse du style : 
2022-02-11 12:06:43 Abeille ERREUR: Abeille1, Message corrompu (taille payload incorrecte, taille=18B, att=9B)
2022-02-11 12:06:41 Abeille ERREUR: Abeille2, Message corrompu (taille payload incorrecte, taille=30B, att=26B)
2022-02-11 12:06:41 Abeille ERREUR: Abeille1, Message corrompu (taille payload incorrecte, taille=12B, att=1B)
2022-02-11 12:06:30 Abeille ERREUR: Abeille2, Message corrompu (taille payload incorrecte, taille=32B, att=22B)
2022-02-11 12:06:26 Abeille ERREUR: Abeille1, Message corrompu (taille payload incorrecte, taille=23B, att=22B)
2022-02-11 12:06:24 Abeille ERREUR: Abeille2, Message corrompu (taille payload incorrecte, taille=17B, att=9B)

Après désactivation de la zigateV1 donc abeille2, plus d'erreur.

Je te mets les log. si vraiment tu veux qu'on creuse, on peut, mais en partant du principe qu'on s'était déjà bien prit la tête la dessus la dernière fois, faut pas non plus y passer trop de temps. Si tu vois rien d'intéressant on s'embète pas.
Pour moi, c'est lié à la zigate et pas à abeille, donc si tu vois rien, je pense pas que ca soit utile
[AbeilleLogs-220211 (1).tar.gz](https://github.com/fairecasoimeme/ZiGate/files/8048101/AbeilleLogs-220211.1.tar.gz)
.

tcharp38, 2022-02-11:
Merci @Jake309
Dans ton cas il y a des tas d erreurs CRC, qui pour un device USB n'est pas normal.

- Soit le port ttyUSB0 est parfois en conflit (2 utilisateurs, Abeille et ??)
- Soit tu as peut etre un souci avec le convertisseur TTL.

J'ai cru voir qqch de @pipiche38 la dessus recemment.

Jake309, 2022-02-11:
J'ai des erreurs a pas mal d'endroit dant le les log cmd.
J'ai refais une inclusion d'une prise qui me sert pas, et après réintégration, elle réagit. 
J'espère que je ne vais pas devoir réinclure tout :/ 
Elle est vraiment possédé cette zigatev1 !

[AbeilleLogs-220211 (2).tar.gz](https://github.com/fairecasoimeme/ZiGate/files/8048512/AbeilleLogs-220211.2.tar.gz)

tcharp38, 2022-02-11:
Jake.
Suis un peu perdu. Je crois qu'il y a des soucis Abeille par dessus ces essais alors on va pas pourrir ce forum.

Finalement ta v1 seule, je ne sais pas quel est son status. Mais compliqué si tu n'as pas  moyen de tester en dehors de ton serveur. Si seule, elle a des soucis de CRC, elle, ou plutot son convertisseur TTL est a changer, et ca concerne ce forum.

Pour les souci multi-zigate, je viens d'ouvrir ce sujet https://github.com/KiwiHC16/Abeille/issues/2323

pipiche38, 2022-02-11:
Attention quand on flash un firmware optiPdm alors qu'on était en legacy il edt critique de bien faire un Erase EEPROM lors du flash

Jake309, 2022-04-11:
C'est à dire "il est critique de bien faire" ? 
Tu dis qu'il faut le faire, on se comprend bien. ?! 
Dans ce cas on est d'accord. 
J'ai suivi pas a pas la procédure sur le site zigate. C'était pas mon premier flash donc je m'attendais pas à ce résultat. 
Je dois envoyer un module qui fait des choses bizarre sur la v2, je vais demander si je peux y mettre la zigate v1 avec

Jake309, 2022-08-30:
Je fais un déterrage parce que ca m'agace un peu mon problème avec ma V1.
Il est toujours d'actu.
J'ai même flashé sous linux pour tester au cas ou il y aurait un bug via windows, rien ne change.

Avec testGUI j'ai toujours la meme chose que sur mon 1er poste.
Dans abeille, on voit que le plugin attend une réponse de la zigate:
https://github.com/KiwiHC16/Abeille/issues/2432

@fairecasoimeme quand je t'ai envoyé mon orvibo pour détecter le bug, tu l'avais testé et dis qu'elle fonctionnait.
Pourrais tu m'aider stp ?

Merci

fairecasoimeme, 2022-09-01:
Bonjour,
Ça remonte à pas mal de temps mais il me semble que oui tout fonctionnait.
Si vous avez fait un backup, reprenez les tests avec testGUI en ayant fait un erase EEprom de la ZiGate.