fairecasoimeme/ZiGate #394: Error: Zigate: Decode8000 Zigate v1 Usb

blooesky, 2022-01-20:
Hi,

After 3.xx firmware update i have an error which freezes the Zigate v1 Usb ttl.

Error: Zigate: Decode8000 - PacketType: 0092 TypeSqn: 01 sqn_app: 00 sqn_aps: bb Status: [0a] 

Domoticz 2021.1(build 14081) Zigate plugin 6 beta , zigate firmware 3.21

Thank you!

waltervl, 2022-02-24:
I have the same issue, see https://www.domoticz.com/forum/viewtopic.php?p=287236#p287236
zigate firmware 3.21

Tibothika, 2022-02-25:
I also have the same issue:

Decode8000 - PacketType: 0092 TypeSqn: 01 sqn_app: 1c sqn_aps: da Status:[E_ZCL_ERR_ZTRANSMIT_FAIL]

Domoticz Beta 2022.1 (build 14205), PiZigate V1 plugin beta (build 6.0.114), firmware 3.21

pipiche38, 2022-09-11:
in such case, the best is to simply stop the plugin. un-plug the Zigate. wait for 1 minute and then plug-in again, and restart