fairecasoimeme/ZiGate #396: mise à jour Pizigate impossible

Jerome63000, 2022-04-01:
Bonjour à tous,
J'ai une Pizigate V1 montée dans un Pi3B+. Tout fonctionnait parfaitement jusqu’à ce mon Alarme Jeedom ne fonctionne plus (Firmware Zigate trop ancien pour le plugin Abeille). Du coup j'ai tenté de faire la mise à jour, et c'est l’échec total, impossible.
Aujourd'hui ma Pizigate reste en led bleu faible.

Lorsque je fais la commande :

sudo nano /boot/cmdline.txt

J'ai ceci :

console=tty1 root=PARTUUID=21bf9052-02 rootfstype=ext4 elevator=deadline fsck.r$

Normal ou il y a un soucis?

Et lorsque je test la Pizigate voici ce que ça donne :

pi@jeedom:~/PiZiGate-tools/test $ sudo ./PiZiGate_test /dev/ttyAMA0
RPI Model searching ...
Warning !!! you must modify /boot/config.txt
Verifying /boot/cmdline.txt ...
+ /boot/cmdline.txt seems to be OK
Verifying /boot/config.txt ...
+ /boot/config.txt seems to be OK
Warning !!! if it's not the case, you have to execute the following commands :
sudo systemctl disable hciuart
sudo usermod -aG gpio pi
then reboot the PI
Searching /dev/ttyAMA0...
+ Port : /dev/ttyAMA0 is not used
Verif GPIOs ...
+ GPIO 0 (RESET) --> OK
+ GPIO 2 (FLASH) --> OK
Config GPIOs ...
+ GPIO 0 (RESET) --> OK
+ GPIO 2 (FLASH) --> OK
Opening : /dev/ttyAMA0 ...
+ /dev/ttyAMA0 opened --> OK
+ Packet 01 02 10 10 02 10 02 10 10 03 sent --> OK
- No packet received - size sent : 11 - error : 11

Sinon quand j'essai de reflasher , à l'étape "make" j'ai cette erreur :

pi@jeedom:~/JennicModuleProgrammer/Build $ make
make: Nothing to be done for 'all'.

Idem avec "sudo make" :

pi@jeedom:~/JennicModuleProgrammer/Build $ sudo make
make: Nothing to be done for 'all'.

et quand je lance "wget https://github.com/fairecasoimeme/ZiGate/releases/download/Release/ZiGate_v30f.bin" j'ai l'erreur :

ZiGate_v30f.bin: Permission denied

Cannot write to ‘ZiGate_v30f.bin’ (Permission denied).

Help... Merci d'avance (pour info je suis novice en SSH, j'ai beaucoup suivi des tuto)

tcharp38, 2022-04-01:
Salut. Si tu utilises Abeille tu peux flasher la pizigate directement de la page de config.
A tenter.

Autre point la version à prendre est la 3.21 et surtout pas la 3.0f dont tu parles.
Tu es en quelle version actuellement ?

Jerome63000, 2022-04-01:
J'ai essayé en premier via abeille, sans succès. 
Après je ne sais combien de test et d'essais je suis tombé sur ce post, et en appliquant ce qu'il y avait à l'instant, la mise à jour a fonctionné, incompréhensible... 
https://zigate.fr/forum/topic/liste-commandes-maj-firmware-pizigate-v1-raspberry-v3/
Je suis donc maintenant en version 3.1e, mais j'ai peur de tenter la màj pour la version 3.21 OPDM...
Si je tente via Abeille tout se fait tout seul ou il faut que je désactive le demon ou mette en flash la Pizigate, ou quelconque autre manip ?

tcharp38, 2022-04-01:
Tu es parti de quelle version avant d'arriver en 3.1e ? C est tres important.

Cote Abeille, tout se fait tout seul pour la PiZigate mais tu ne peux pas choisir n importe quel FW sans savoir ou tu vas.
Typiquement si ta version initiale etait anterieure a 3.0f tu n'as PAS le choix que de faire un erase PDM et donc TOUT reappairer.
Auquel cas, profite en pour passer en 3.21-OPDM.

Si tu avais une version >= 3.0f, et que tu ne veux pas repartir de zero, la 3.21-LEGACY est celle qu'il te faut. Inutile de rester sur la 3.1e.

Jerome63000, 2022-04-01:
J'avais une version antérieure à la 3.0, mais ça n'avait pas fonctionné, j'avais un message d'erreur en essayant d'installer la version actuelle 3.1e. J'essaierai de faire la mise à jour 3.21 OPDM en croisant les doigt... 
Maintenant j'ai toujours le soucis de mon plugin Alarme qui ne fonctionne toujours pas et je ne sais pas pourquoi...

Jerome63000, 2022-04-01:
J'ai une erreur quand je lance le demon Abeille :
ERREUR: Abeille1, Message corrompu (taille payload incorrecte, taille=14B, att=9216B)

tcharp38, 2022-04-01:
Pour le plugin Alarme c'est hors sujet. Je ne connais pas et rien à voir avec la Zigate.

Mais.. pour la version tu avais une antiquité si < 3.0f. Du coup désolé... pas le choix que 
- d'effacer l'EEPROM (PDM)
- refaire un appairage complet.
- ET du coup du passes en version OPDM donc **3.21-OPDM** comme ca tu aura la toute derniere mise à jour.

Jerome63000, 2022-04-01:
Donc là je suis en 3.1e, j'ai essayé de faire la mise à jour 3.21OPDM depuis Jeedom plugin Abeille et j'ai un message d'erreur :

http://xxx.xxx.x.xx/core/php/getResource.php?file=core/js/log.class.js&md5=97aae7c356368e6a141e894799c00e88&lang=fr_FR | 246 | Uncaught ReferenceError: $rawLogCheck is not defined

Que fait cette manip d'effacer l'EEPROM? Pas de danger pour la Pizigate? Je refaire des manip pour la réinstaller dans mon RPi?
Comment je fais pour effacer l'EEPROM? 
Une fois effacé je retente une mise à jour avec ces commandes et le bon firmware?
https://zigate.fr/forum/topic/liste-commandes-maj-firmware-pizigate-v1-raspberry-v3/

tcharp38, 2022-04-01:
ok tu cumules les soucis.

Effacer l'EEPROM (ou PDM) consiste à effacer le reseau memorisé dans la zigate. Obligatoire car format interne different.
Donc sans danger mais ca impose de reappairer tes equipements qui ne seront plus dans aucun reseau.

On bascule sur un sujet specifique Abeille: https://github.com/KiwiHC16/Abeille/issues/2369
L'erreur que tu mentionnes est deja fixée mais visiblement que dans la beta. Bascule dessus STP.
On se retrouve sur l'autre forum. Je n'ai pas les billes "hors Abeille".