fairecasoimeme/ZiGate #399: Pizigate mode routeur

RudySixOw, 2022-05-30:
Bonjour, je possede une pizigate de première generation (il me semble), dont voici les refs: AA00070530 01c61ba5.
J'ai un réseau zigbee qui fonctionne avec zibbee2mqtt et j'aimerai savoir si il est possible d'utiliser ma pizigate en mode routeur.
J'ai cherché un peu d'info sur d’éventuels firmware mais je n'ai pas trouvé.
Merci d'avance si une âme charitable à une info et désolé si c'est question a déjà été posée.

pipiche38, 2022-05-30:
https://github.com/badzz/Zigate-Router

RudySixOw, 2022-05-30:
Merci pour l'info je regarde mais pour le moment je ne comprends pas trop la procédure pour y parvenir.

RudySixOw, 2022-05-30:
J'ai flashé avec Router_JN5168_DR1175.bin sans vraiment être sûr que mon chipset soit celui-ci. Le flash c'est bien passé mais je ne vois pas de procédure pour le pairring et je ne vois pas d'annonce sur le reseau zeigbee.

ruimarinho, 2022-05-30:
I haven't had much luck with the router firmware as ZHA doesn't seem to like the way it reports. Have you @pipiche38 ?

pipiche38, 2022-05-30:
Je crois que la Zigate V1 etait sur la base d'un JN5168.

A ma connaissance , le mode d'appairage, c'est 

mettre le reseau en mode appairage, puis brancher la zigate-router.

Si tu souhaites faire un reset et refaire l'appairage il faut reflasher

Il faut bien penser à faire un FULL Erase EEPROM lors du flash

ruimarinho, 2022-05-30:
Yep, I've done that but ZHA doesn't really know how to handle its messages. I guess I'll need to work on a quirk.

RudySixOw, 2022-05-31:
J'ai bien fait un erase lors du flash en tout cas je l'ai vu passer. A priori tout c'est bien passé mais dans les logs de zigbee2mqtt que j'utilise avec une trentaine de périphériques zigbee je ne vois aucune trace du routeur pizigate v1.

RudySixOw, 2022-05-31:
Mince j'ai du louper un truc dans la procedure je vais regarder. Merci pour l'info je te tiens au courant.

RudySixOw, 2022-05-31:
J'ai tapé ça en commande et il me semble que le -e c'est bien erase:
./JennicModuleProgrammer -V 6 -P 115200 -e -f Router_JN5168_DR1175.bin -s /dev/ttyAMA0

Ce n'est pas celui-ci dont tu parles ?

pipiche38, 2022-09-23:
oui c'est ca