fairecasoimeme/ZiGate #401: Constant Status.NoFreeAPSAckHandles from zigpy_zigate

Rudd-O, 2022-08-22:
I am using the original ZiGate device that comes with the semitransparent black case.  My network contains roughly 35 devices, with many devices acting as routers because many are connected to mains power.

I am seeing this constantly from the Home Assistant logs:

```
Aug 22 01:47:16 assistant hass[320389]: 2022-08-22 01:47:16.904 WARNING (MainThread) [zigpy_zigate.zigbee.application] Extended error code Status.NoFreeAPSAckHandles
```

Often commands sent to my devices will simply just fail when this is happening:

```
Aug 22 01:47:56 assistant hass[320389]: 2022-08-22 01:47:56.112 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x4120:1:0x0b04]: failed to get attributes '['active_power', 'rms_current', 'rms_voltage']' on 'electrical_measurement' cluster: 
...
Aug 22 01:15:14 assistant hass[318231]: 2022-08-22 01:15:14.948 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xF3BC:1:0x0006]: command failed: 'on' args: '()' kwargs '{}' exception: '[0xf3bc:1:0x0006]: Message send failure'
Aug 22 01:15:14 assistant hass[318231]: 2022-08-22 01:15:14.948 DEBUG (MainThread) [homeassistant.components.zha.entity] light.gym_ceiling_light_on_off: starting transitioning timer for 1.25
Aug 22 01:15:14 assistant hass[318231]: 2022-08-22 01:15:14.948 DEBUG (MainThread) [homeassistant.components.zha.entity] light.gym_ceiling_light_on_off: turned on: {'on_off': DeliveryError('[0xf3bc:1:0x0006]: Message send failure')}
```

This is for a device which I can see in the network route chart to be *directly connected* to my ZiGate.  The ZiGate has all 16 device slots (I think it supports) used.

What can I do to increase these mysterious "APS acknowledgement handles" in the device?

Thanks.

pitchoun38, 2022-12-08:
This is due to the fact that
(1)  zigpy_zigate dosn't respect the Zigate protocol and send zigate command despite the fact that the previous one has not be fully sent and received.  ( https://github.com/zigpy/zigpy-zigate/issues/120 )

(2) that you have some zigbee devices which are sending messages with ACK and which are exhausting the APS buffers in the zigate which are unfortunatly not controlled correctly .

Rudd-O, 2022-12-11:
I threw the device in the trash.  Running a ZigStar device now.  Unforgivable user experience with ZiGate — totally destroyed my ZigBee network and had to start from scratch with a hundred devices — just the electrician's bill to re-pair the devices he had installed in wall boxes was hundreds of dollars.

bokub, 2024-01-16:
Everything was working fine for me during more than a year, but now I constantly see these kind of error messages, and my devices randomly disconnect from my Zigbee network (I only have 13 devices !).

I have to re-pair them manually multiples times a week.

```
Logger: zigpy_zigate.zigbee.application
Source: runner.py:188
Unexpected message send failure
Extended error code Status.NoFreeAPSAckHandles
Extended error code Status.ResourceShortage
```

Is there any way to fix this? Or should I throw my device in the trash too?

Rudd-O, 2024-01-20:
Trash.

Rudd-O, 2024-01-20:
I'm closing this issue, nobody is going to fix this.

closed 2024-01-20 by Rudd-O