fairecasoimeme/ZiGate #403: [Question]  Is windowCovering command only available for 3.0f or also in later versions

dmsherazi, 2022-09-22:
The Dcumentation here says only 3.0f
[image]

pipiche38, 2022-09-23:
yes it is available upwards. but you can also re-implement them using the 0x0530 command which allow to by-pass the all NXP stack.
You have also zigpy project which implement a full stack to controll the coordinator in a radio hardware indepence.

dmsherazi, 2022-09-29:
Tested and Works. Although There is a small bug in the TEST GUI linked here 
 -> https://github.com/fairecasoimeme/ZiGate/issues/125#issuecomment-1255016373
 Bug -> The Extra command param must be a u16 and not u8

closed 2022-09-29 by dmsherazi