fairecasoimeme/ZiGate #406: Help & guidance in implementing the Tuya FE00 cluster for TS0601 devices

dmsherazi, 2022-11-01:
I am trying to add a thermostat and some curtains that are implemented by the tuya generaic devies with cluster FE00.
Are the FE00 clusters commands supported? Kindly guide me in how to implement them for me to get started!

tcharp38, 2022-11-05:
Hello @dmsherazi 
I can just give you my point of view:
- I assume you are talking about EF00 and not FE00 but anyway would be the same answer
- EF00 is private cluster. Therefore DOES NOT follow any Zigbee standard. Tuya does what it wants
- Could not be supported by Zigate since there is no spec
- Your plugin has to handle this itself by decoding it and doing custom stuff
- You may find useful infos there: https://developer.tuya.com/en/docs/iot/tuya-zigbee-universal-docking-access-standard?id=K9ik6zvofpzql

Supporting private cluster is really time consuming.
Good luck.

tcharp38, 2022-12-12:
Hi
Please refer to tuya.developer link I put above and specifically this https://developer.tuya.com/en/docs/iot/tuya-zigbee-universal-docking-access-standard?id=K9ik6zvofpzql#subtitle-5-Private%20cluster

Not sure about attributes but I see mainly commands sent or received by Tuya device and containing 1 to several datapoints.

You can also look into Abeille's code to see how it is currently handled but.. not a reference.

dmsherazi, 2022-12-13:
> You can also look into Abeille's code to see how it is currently handled but.. not a reference.
Can you provide a link?

I see that zigpy-zigate works with zigate and they support tuya devices reading the DP and sending commands. Is there any serial commands for zigate to request it to send the custom tuya commands ?
Looking at  this list [](https://zigate.fr/documentation/commandes-zigate/) I dont see anything except **Requête APS**
[image]
that can send any custom commands? Do I have to use this to send the tuya commands?

tcharp38, 2022-12-13:
I can only talk how it is currently handled in Abeille but as I said probably not the cleanest code
So yes, Abeille is using 0530 too

https://github.com/KiwiHC16/Abeille/blob/dd6367914a0a3314867c76794ee3375b1a284e38/core/class/AbeilleCmdProcess.class.php#L4517
and
https://github.com/KiwiHC16/Abeille/blob/master/core/php/AbeilleCmd.php