fairecasoimeme/ZiGate #407: FW 0004-0321: 804E may return wrong info => use 8002 instead

tcharp38, 2022-11-04:
Seen with FW OPDM 3.21, "hybrid" mode on Abeille plugin

Summary: The 8002 message show invalid datas but the 804E gives datas anyway... so wrong datas

I suggest to no longer use 804E and simply decode 8002 to detect such behavior.
At least.. this is now the way Abeille will handle LQI response.

    [2022-11-04 07:10:34] Got "80020015340000008031000002000002000014000F000B020000"
        8002
        0015
        34
            00
            0000
            8031
            00
            00
            02
            0000
            02
            0000
                14 => sqn
                00 => status
                0F => neighbor table entries
                00 => index
                0B => neighbor table list count
                0200 => INVALID LIST !!!
        00
        
        [2022-11-04 07:10:34] Abeille1, Type=8002/Data indication, Status=00, ProfId=0000, ClustId=8031, SrcEP=00, DstEP=00, SrcAddrMode=02, SrcAddr=0000, DstAddrMode=02, DstAddr=0000
        [2022-11-04 07:10:34]   SQN=14, Status=00, NTableEntries=0F, startIdx=00, nTableListCount=0B
        [2022-11-04 07:10:34]   Handled by decode804E
        
    [2022-11-04 07:10:34] Got "804E00EFA714000F0B000002CD52685D02A63DB100158D00040018005D686ACD5200000000000AE53500158D00ED5A77112502AB8D000400015004007F60CD52685D00000000174351140052AD700100000000803114000400036C0B0F00686A52685D02A63DB100158D00040018005D52CD000004000AE53500158D00ED5A77112502AB8D010050700160CD52685D00000000174351140052AD0000003DB114000400036C0B0F00686A52685D02A61500C1ED5A18005D52CD000004000AE53500158D001177695114010050700160CD52685D0000000017435200E102A63DB114000400036C0B0F00686A52685D001545000000"

        804E
        00EF
        A7
            14 => sqn
            00 => status
            0F => neighbor table entries
            0B => neighbor table list count
            00 => index
                0002 => bullshit !!
                CD52685D02A63DB1 => bullshit !!
                00158D0004001800 => bullshit !!
                5D => bullshit !!
                68 => bullshit !!
                6A => bullshit !!
                The rest is bullshit too.
                CD5200000000000AE53500158D00ED5A77112502AB8D000400015004007F60CD52685D00000000174351140052AD700100000000803114000400036C0B0F00686A52685D02A63DB100158D00040018005D52CD000004000AE53500158D00ED5A77112502AB8D010050700160CD52685D00000000174351140052AD0000003DB114000400036C0B0F00686A52685D02A61500C1ED5A18005D52CD000004000AE53500158D001177695114010050700160CD52685D0000000017435200E102A63DB114000400036C0B0F00686A52685D001545000000
                
        [2022-11-04 07:10:34] Abeille1, Type=804E/Management LQI response, SQN=14, Status=00, NTableEntries=0F, NTableListCount=0B, StartIndex=00, SrcAddr=0000
        [2022-11-04 07:10:34]   NAddr=0002, NExtPANId=CD52685D02A63DB1, NExtAddr=00158D0004001800, NDepth=5D, NLQI=68, NBitMap=6A => RxONWhenIdle/Sibling/?/EndDevice

pitchoun38, 2022-11-24:
Hybrid mode was put in place for discovery and analysis of new devices, it was never in our mind  ( @badzz , nor myself) to have a production running in hybrid mode, as you are getting duplicate messages

tcharp38, 2022-11-24:
I don't care about duplicated messages but I care about wrong infos.
804E is definitively not reliable if the incoming packet is wrong as shown above.

So if "hybrid" is not "production" level is "raw" mode better ?

pipiche38, 2022-11-24:
on our plugin with zigate, we rely on the standard mode and the 0x804E behaves correctly.
the zigpy library works with the RAW mode , I don't have much feedbacks on that

tcharp38, 2022-11-24:
Ok thanks. 
I'll add to test raw mode before leaving hydrid.