fairecasoimeme/ZiGate #410: zigate: Error reading module information - check cabling and power while upgrading firmware

karawitan, 2024-02-05:
Hello,

  I have a Zigate USB, whose firmware I'm trying to upgrade:
I've tried different baudrates, but only 38400 seems applicable. All other values report : "Error setting baudrate" 

Now, here is the command I type which goes a little bit further:

```
$ sudo ./JennicModuleProgrammer -V 6 -e -P 38400 -f FlashProgrammerExtension_JN5168.bin -s /dev/ttyUSB0
JennicModuleProgrammer Version: 0.7(r56124)
Error reading module information - check cabling and power
```

Below some more informations about my setup:

On kernel side, the pl2303 is loadded fine,

```
[   21.858068] usbcore: registered new interface driver usbserial_generic
[   21.859428] usbserial: USB Serial support registered for generic
[   21.870022] usbcore: registered new interface driver pl2303
[   21.871255] usbserial: USB Serial support registered for pl2303
[   21.872401] pl2303 4-1:1.0: pl2303 converter detected
[   22.093699] usb 4-1: pl2303 converter now attached to ttyUSB0

$ lsmod | grep pl2
pl2303                 32768  0
usbserial              73728  1 pl2303
usbcore               409600  10 pl2303,usbserial,xhci_hcd,ohci_hcd,ehci_pci,usbhid,ehci_hcd,btusb,xhci_pci,ohci_pci
```

Checking current firmware version is working also

```
python3 zigate-list-version.py
INFO:zigate:Searching ZiGate port
INFO:zigate:ZiGate found at /dev/ttyUSB0
OrderedDict([('major', 3), ('installer', '321'), ('lqi', 0), ('version', '3.21')])
3.21
WARNING:zigate:Unknown response 0x8012
True
[]
```

Do you have any clues about what may be going wrong ?

tcharp38, 2024-02-06:
Hello @karawitan 
Forget DK6. I think it is for Zigate v2/+ based on JN5189

Your Zigate is v1 with old FW (legacy).
I'm afraid ttyUSB0 is not a "standard" serial port and therefore Jennic programmer may not work. I only use it for my PiZigates.
For USB I would suggest the windows application that you can find on Zigate site.

karawitan, 2024-02-06:
Hi @tcharp38 , thank you very much for the feedback.

Ok let's forget DK6

I don't think ttyUSB0 may influence in any case the behaviour of Jennic programmer, given the way linux works. Moreover, documentation from zigate.fr uses ttyUSB0 too :)

By the way, as the chip is MS5168-M05, I will try using nxp-isp that is available on linux.