fairecasoimeme/ZiGate #411: mise à jour impossible

toto6976, 2024-09-08:
Bonjour,

J'ai une pizigate v1 (je pense) installée sur une dietpi v.9. Je souhaiterais faire la mise à jour sur le dernier firmware Zigate V3.23 (opdm) ou (legacy);
J'ai l'erreur suivante : 
```
dietpi@DietPi-Salon:~/JennicModuleProgrammer/Build$ sudo ./JennicModuleProgrammer -V 6 -P 115200 -f ZiGate_v3.23-legacy.bin -s /dev/ttyAMA0
JennicModuleProgrammer Version: 0.6 (r56124)
Setting baudrate: 115200
Error setting baudrate 
```

Pouvez-vous m'aider ?

tcharp38, 2024-09-08:
Salut @toto6976 
J'ai déja vu passer ce genre d'erreur de mon coté et 2 choses me reviennent en tete
- soit un souci au niveau kernel avec l'UART utilisée par la PiZigate. A voir peut etre dans "dmesg"
- soit la PiZigate n'est pas en mode "flash". Il faut jouer avec les PIOs pour ca comme mentionné dans la doc Zigate

La mise à jour n'est pas prise en compte dans l'environnement dans lequel tu utilises la Zigate ? (Domoticz, Abeille..)

toto6976, 2024-09-08:
La mise à jour n'est pas prise en compte dans zha.
J'ai configuré comme décrit dans la doc. Ce qui est bizarre c'est que lorsque je passe la commande 'gpio write 0 0' la led se met en bleu faible. Et lorsque je lui passe 'gpio write 0 1' comme décrit dans la doc elle repasse en  bleu fort.

toto6976, 2024-09-08:
J ai trouvé un poste dans lequel quelqu'un a rencontré des erreurs similaires aux miennes et à réussi la maj avec un tuto. Problème, le site a dû fermer car il n y a rien en ligne. Aurais tu par hasard la liste des commandes citées ci-dessous ?
https://zigate.fr/forum/topic/liste-commandes-maj-firmware-pizigate-v1-raspberry-v3/

tcharp38, 2024-09-08:
Mais la Zigate est reconnue dans ZHA et fonctionnelle ?
Si non, ca confirme un souci de controle de l UART, probablement utilisée par qqch d'autre.

toto6976, 2024-09-08:
La zigate est reconnue par ZHA.
Fonctionnelle je ne sais pas : j'essaie en vain d'appairer un capteur de
mouvement sonoff. Zha le reconnaît mais je ne le vois nulle part après !?!

toto6976, 2024-09-08:
Pour ce qui est de la configuration j'ai un peu galéré car les commandes ne
sont plus les mêmes sous debian.
Donc y'a peut être un truc qui n'est pas comme il faut !

toto6976, 2024-09-08:
Voilà le retour du test:

***@***.***:~/PiZiGate-tools/test$ sudo ./PiZiGate_test /dev/ttyAMA0
RPI Model searching ...
Warning !!! you must modify /boot/config.txt
Verifying /boot/cmdline.txt ...
+ /boot/cmdline.txt seems to be OK
Verifying /boot/config.txt ...
+ /boot/config.txt seems to be OK
Warning !!! if it's not the case, you have to execute the following
commands :
sudo systemctl disable hciuart
sudo usermod -aG gpio pi
then reboot the PI
Searching /dev/ttyAMA0...
sh: 1: lsof: not found
+ Port : /dev/ttyAMA0 is not used
Verif GPIOs ...
+ GPIO 0 (RESET) --> OK
+ GPIO 2 (FLASH) --> OK
Config GPIOs ...
+ GPIO 0 (RESET) --> OK
+ GPIO 2 (FLASH) --> OK
Opening : /dev/ttyAMA0 ...
+ /dev/ttyAMA0 opened --> OK
+ Packet 01 02 10 10 02 10 02 10 10 03 sent --> OK
- No packet received - size sent : 11 - error : 11

tcharp38, 2024-09-08:
Les GPIOs semblent controlables et ok mais la Zigate ne repond pas du tout.
Elle a déja fonctionné auparavent ?
Si oui ca semble confirmer un souci de port ttyAMA0. Il faut que personne ne le lise pendant ce test.

toto6976, 2024-09-08:
Oui, elle a fonctionné sur une autre installation home assistant (core).
Après reboot, j'ai ça 
./PiZiGate_test /dev/ttyAMA0
RPI Model searching ...
Warning !!! you must modify /boot/config.txt
Verifying /boot/cmdline.txt ...
+ /boot/cmdline.txt seems to be OK
Verifying /boot/config.txt ...
+ /boot/config.txt seems to be OK
Warning !!! if it's not the case, you have to execute the following commands :
sudo systemctl disable hciuart
sudo usermod -aG gpio pi
then reboot the PI
Searching /dev/ttyAMA0...
sh: 1: lsof: not found
+ Port : /dev/ttyAMA0 is not used
Verif GPIOs ...
+ GPIO 0 (RESET) --> OK
+ GPIO 2 (FLASH) --> OK
Config GPIOs ...
+ GPIO 0 (RESET) --> OK
+ GPIO 2 (FLASH) --> OK
Opening : /dev/ttyAMA0 ...
+ /dev/ttyAMA0 opened --> OK
+ Packet 01 02 10 10 02 10 02 10 10 03 sent --> OK
+ Packet received --> OK
size : 82
 01 00 80 24 00 0d a4 01 00 00 00 15 8d 00 01 ed f8 8b 0b 00 03
01 80 06 00 02 85 01 00 03
01 80 00 00 05 95 00 00 00 10 00 03
01 80 10 00 05 8f 00 03 03 1a 00 03

tcharp38, 2024-09-08:
Ha cette fois le test est sucessful.
Tu configure les PIO pour le mode flash et retente le flashage ?

toto6976, 2024-09-08:
Je veux bien. Tu valides les étapes à chaque fois ?

toto6976, 2024-09-08:
Gpio mode 0 out
Gpio mode 2 out
Gpio write 2 0
Gpio write 0 0
Gpio write 0 1

tcharp38, 2024-09-08:
Ca semble correct
Que dit l etape suivante du coup?

toto6976, 2024-09-08:
Cd JennicModuleProgrammer
Build /
sudo make créant
Sudo make all

tcharp38, 2024-09-08:
Je ne pige pas ces etapes de compile mais bon tu avais déja le binaire du programmeur.
Tu l'as lancé pour voir ?

toto6976, 2024-09-08:
sudo ./JennicModuleProgrammer -V 6 -P 115200 -f Zigate_v3.23-legacy.bin -s /dev/ttyAMA0

toto6976, 2024-09-08:
Setting baudrate 115200
error setting baudrate

tcharp38, 2024-09-08:
Zut :(
Je vais essayer de retrouver des infos. J'ai deja eu ca mais je ne me souviens pas comment je m'en suis sorti.

En tout cas ma derniere hypothese est plutot vers le programmeur.. peut etre plus compatible avec ta distrib.

toto6976, 2024-09-08:
La pizigate+ (v2) fonctionnerait ?

toto6976, 2024-09-08:
OK. Bon je vais me coucher. J'attends de tes nouvelles. 

Merci

tcharp38, 2024-09-08:
Je te laisserai retenter par curiosité en supprimant l'option "-P"

sudo ./JennicModuleProgrammer -V 6 -f Zigate_v3.23-legacy.bin -s /dev/ttyAMA0

toto6976, 2024-09-09:
Bonjour,
J'essaie. C'est quoi l'option -P ?

tcharp38, 2024-09-09:
Salut
Le baudrate correspond à la vitesse de transfert.

Mais bon on n'a pas avancé du coup.
Ta distrib est quoi ?

toto6976, 2024-09-09:
Dietpi v9.7 (la dernière). C'est une dérivée de debian 12

tcharp38, 2024-09-09:
Ok donc le code du programmateur n'est probablement plus compatible.
Comme je risque d'avoir le souci bientot je vais essayer de mon coté de migrer vers qqch de + recent que ma buster/v10.

toto6976, 2024-09-09:
Dans ZHA, la pizigate est reconnue il me semble :+1: 

PiZiGate
par ZiGate
Micrologiciel : 3.1a 
IEEE: 00:15:8d:00:01:ed:f8:8b
Nwk: 0x0000
Device Type: Coordinator
LQI: Inconnu
RSSI: Inconnu
Dernière vue: 2024-09-09T1
Source d'énergie: Mains0:01:02

tcharp38, 2024-09-09:
Oui tres bien. C'est en ligne avec le resultat de ton test.

toto6976, 2024-09-09:
Après; le firmware est en 3.1 çà doit passer avec les appareils zigbee. Mon souci majeur, en dehors de la mise à jour, est que l'appareil que je veux appérer n'apparait pas dans les intégrations ZHA alors que l'appairage semble se faire en partie. Je dis en partie car à la fin je n'ai pas la fenêtre dans laquelle il me demande 'la pièce dans laquelle mettre le capteur de mouvement'. 
Mais c'est peut-être le capteur qui déconne car sonoff semble ne pas l'appairer non plus...
Et comme je débute, je n'ai pas d'autres zigbee sous la main

tcharp38, 2024-09-09:
3.1a est une antiquité.
Si tu refais qqch de propre tu dois partir en version OPDM 3.23 donc ca fait du sens de faire la mise à jour maintenant.
Je fais des essais de mon coté

toto6976, 2024-09-09:
Quand j'essaie la version OPDM  3.23, je dois faire l'erase de l'eeprom. Or la valeur -e (pour erase je suppose) dans la commande n'est pas reconnue : 
./JennicModuleProgrammer -V 6 -e -P 115200 -f "<chemin vers le bin>" -s /dev/ttyAMA0

toto6976, 2024-09-09:
> 3.1a est une antiquité. Si tu refais qqch de propre tu dois partir en version OPDM 3.23 donc ca fait du sens de faire la mise à jour maintenant. Je fais des essais de mon coté

Home Assistant supervised a détécté la vetusté du firmware car il m'a proposé une mise à jour lors de l'intégration de ZHA. Je n'ai pas voulu la faire car il me semblait avoir lu qqpart que la Pizigate ne devait être mise à jour que via le logiciel du programmateur.

toto6976, 2024-09-09:
> 3.1a est une antiquité. Si tu refais qqch de propre tu dois partir en version OPDM 3.23 donc ca fait du sens de faire la mise à jour maintenant. Je fais des essais de mon coté

ok. Merci de ton aide.

tcharp38, 2024-09-09:
> Home Assistant supervised a détécté la vetusté du firmware car il m'a proposé une mise à jour lors de l'intégration de ZHA. Je n'ai pas voulu la faire car il me semblait avoir lu qqpart que la [...]

Peut etre que ce programmateur est intégré dans ZHA justement.
A suivre

toto6976, 2024-09-09:
Oui, mais je ne sais pas comment le rappeler !

toto6976, 2024-09-09:
En cherchant sur le forum de home assistant je tombe la-dessus : https://www.home-assistant.io/integrations/zha/
et je lis çà dans le paragraphe concernant la 'zigate' : 
Autres adaptateurs ou modules radio zigbee pris en charge mais non recommandés (bon, c'est pas rassurant !)
Radios basées sur les ziGate (par l'intermédiaire de la[zigpy-zigate](https://github.com/zigpy/zigpy-zigate)bibliothèque de zigpy et nécessite un micrologiciel **3.1d** ou plus tard). Mon firmware est le 3.1a. Donc déjà, çà ne peut pas aller !

tcharp38, 2024-09-12:
Salut
Bon j'ai bataillé a m'installer une debian 12.. et au final pas de soucis pour compiler et flasher ma PIv1.

Avant de lancer ton "JennicModuleProgrammer" comment controles tu les GPIOs ?

toto6976, 2024-09-12:
Bonjour
Avec le commande gpio .

tcharp38, 2024-09-12:
J'ai bataillé car je suis sur une Debian12 (Armbian) pour Orange pi Zero.
Il manquait pas mal de choses et evidement il y a du changement aussi.

Bref ta sequence GPIO est correcte.
Le test arrive bien à lire la version du FW.
Donc le souci est au niveau du programmateur.

C'est toi qui l'as compilé ou tu as récupéré un binaire ?

toto6976, 2024-09-12:
Je ne sais plus : sur github je crois.

toto6976, 2024-09-12:
Tu as un lien ?

tcharp38, 2024-09-12:
Non ma question est "as tu recupéré un source et tu l'as compilé"
ou l'as tu récupéré tel quel executable ?

toto6976, 2024-09-12:
Là il me semble : https://github.com/Jerome-PS/JennicModuleProgrammer

toto6976, 2024-09-12:
J'ai au début essayé avec 'apt' mais je ne dois pas avoir le dépôt dans mon sources.list

toto6976, 2024-09-12:
> Je prepare un petit package avec un script et le code du flasher que j'utilise

Ah...trop bien !! Merci :))

tcharp38, 2024-09-12:
Tiens essaie avec ca.
J'ai mis toute la trace dans le README.txt

[prog_jennic-0.7-Abeille.zip](https://github.com/user-attachments/files/16980947/prog_jennic-0.7-Abeille.zip)

toto6976, 2024-09-12:
OK. Merci

toto6976, 2024-09-12:
en mode $user voilà ce que donne le test :
```
dietpi@DietPi-Salon:~/PiZiGate-tools/test$ ./PiZiGate_test /dev/ttyAMA0
RPI Model searching ...
sh: 1: cannot create model.tmp: Permission denied
Warning !!! you must modify /boot/config.txt
Verifying /boot/cmdline.txt ...                                                                                                                          
sh: 1: cannot create cmdlineTest.tmp: Permission denied
+ /boot/cmdline.txt seems to be OK
Verifying /boot/config.txt ...                                                                                                                           
sh: 1: cannot create configTest1.tmp: Permission denied
sh: 1: cannot create configTest2.tmp: Permission denied
+ /boot/config.txt seems to be OK
Warning !!! if it's not the case, you have to execute the following commands :                                                                           
sudo systemctl disable hciuart                                                                                                                           
sudo usermod -aG gpio pi                                                                                                                                 
then reboot the PI                                                                                                                                       
Searching /dev/ttyAMA0...                                                                                                                                
sh: 1: lsof: not found
sh: 1: cannot create output.tmp: Permission denied
+ Port : /dev/ttyAMA0 is not used
Verif GPIOs ...                                                                                                                                          
+ GPIO 0 (RESET) --> OK
- GPIO 2 (FLASH) --> NOK                                                                                                                                 
Config GPIOs ...                                                                                                                                         
+ GPIO 0 (RESET) --> OK
+ GPIO 2 (FLASH) --> OK                                                                                                                                  
Opening : /dev/ttyAMA0 ...                                                                                                                               
- Error - Unable to open UART.  Ensure it is not in use by another application
```

toto6976, 2024-09-12:
4 réponses dfférentes au test !!! Je ne sais plus quoi penser !?§?

tcharp38, 2024-09-12:
Mais pourquoi tu me montres ça ?

Tu decompresses le package que je t'ai donné
tu fais un cd
make clean all
puis flashZigate.sh ... blablabla

Suivre le README

toto6976, 2024-09-12:
bon...ok

tcharp38, 2024-09-12:
les droits ne sont surement pas bon

chmod +x flashZigate.sh

tcharp38, 2024-09-12:
ATTENTION !!
Ton port n'est pas /dev/ttyS1 il me semble mais /dev/ttyAMA0 non ?

toto6976, 2024-09-12:
> ATTENTION AUSSI !! Le chemin du FW que je donne est un exemple. A toi de mettre le bon

Oh là là !!!!!!!! Je suis fatigué ce soir ! j'ai même oublié de donner les droits !!!

toto6976, 2024-09-12:
```
$ sudo ./flashZigate.sh /dev/ttyAMA0 PI WiringPi ZiGate_v3.23-OPDM.bin
Checking Zigate type PI access on port /dev/ttyAMA0
- /dev/ttyAMA0 port found
= ERROR: Could not find 'lsof' command
```

tcharp38, 2024-09-12:
Home assistant est en train d'utiliser la Zigate.
il faut l'arreter ou seulement le plugin qui gere la Zigate.
Elle doit etre libre pour pouvoir faire une mise à jour.

toto6976, 2024-09-12:
Je ne sais pas.comment faire pour arrêter home assistant

tcharp38, 2024-09-12:
Alors la.. je ne peux pas t'aider. Je ne connais pas cet environnement.
Mais tant qu'il gere la Zigate, aucune chance de la mettre à jour. Il ne peut y avoir qu'un seul controle.

toto6976, 2024-09-12:
Je vais jeter un œil et demander sur le forum de home assistant. Au pire je l'enlève ce sera pas pire je pense.
Je te tiens au courant

toto6976, 2024-09-13:
Bonjour,
Mise à jour ok. Merci pour tout. Je vais voir le reste maintenant.
```
$ sudo ./flashZigate.sh /dev/ttyAMA0 PI WiringPi ZiGate_v3.23-OPDM.bin
Checking Zigate type PI access on port /dev/ttyAMA0
- /dev/ttyAMA0 port found
- Port seems free
Configuring PI Zigate for 'flash' mode (lib=WiringPi)
Flashing Zigate
- Port tty: /dev/ttyAMA0
- Type    : PI
- Lib GPIO: WiringPi
- File    : ZiGate_v3.23-OPDM.bin

JennicModuleProgrammer Version: 0.7-r56124-Abeille
- Detected Chip: JN5168
- MAC Address  : 00:15:8D:00:01:ED:F8:8B
Setting baudrate: 115200
Opened firmware file size 194980
Module uses Bootloader v2 Header
Erasing:   100%
Writing Program to Flash
Writing:   100%
Verifying Program in Flash
Verifying: 100%
= Success
Restoring PI Zigate GPIOs config for prod mode
= Tout s'est bien passé. Vous pouvez fermer ce log.
```

tcharp38, 2024-09-13:
Ha super !
Penses à fermer le sujet du coup.

[3 more replies]

closed 2024-09-13 by toto6976